arduino/reference-en #15: Specify data types in function definitions

nickgammon, 2015-04-25:
I suggest that the types of arguments and return values be specified in the documentation.  For example instead of:

> **Syntax**
>
>    digitalRead(pin)
>
>    **Parameters**
>
>    **Returns**
>
>    HIGH or LOW

State the data types, eg.

> **Syntax**
>
>    **int** digitalRead(**byte** pin)
>
>
>    **Parameters**
>
>    **Returns**
>
>    HIGH or LOW

You shouldn't have to muck around looking in the source just to find if the argument is a byte or an int, or what type the return value is.

Also in this particular case, couldn't "parameters" be explained? eg. 

>    **Parameters**
> 
>   **pin**: The pin number to read from, on the Arduino board. This is probably not the pin number on the actual processor chip.

per1234, 2015-05-05:
Although that is a standard way to indicate data type in documentation I think that for the beginners Arduino is intended to be accessible to this may be a bit confusing because they may think they need to include the int and byte in their code. I think it would be more clear to state type separately like this:
> **Syntax**
>
> digitalRead(pin)
>
> **Parameters**
>
> pin: the number of the digital pin you want to read
>
> * Type: byte
>
> **Returns**
>
> HIGH or LOW
>
> * Type: int

The digitalRead() reference actually does specify the type of the pin parameter:
> pin: the number of the digital pin you want to read (int)

and I think this is a reasonable way to indicate data type but this is not done consistently in other reference pages and the return type is not specified even on the digitalRead() page. Also the type is incorrect, it should be byte not int.

I'd be happy to submit a pull request with the data type specification on all functions but I'd like to get some input on the best format.

nickgammon, 2015-05-06:
Yes, it is a pity the only reference to the type of data is incorrect.

It is also a pity that the example shows **int** which reinforces the error:

```
int ledPin = 13; // LED connected to digital pin 13
int inPin = 7;   // pushbutton connected to digital pin 7
int val = 0;     // variable to store the read value

void setup()
{
  pinMode(ledPin, OUTPUT);      // sets the digital pin 13 as output
  pinMode(inPin, INPUT);      // sets the digital pin 7 as input
}

void loop()
{
  val = digitalRead(inPin);   // read the input pin
  digitalWrite(ledPin, val);    // sets the LED to the button's value
}
```

Your suggested alternative could be OK, although I think you do have to learn **something** to code. I mean, are you going to explain what the word "type" means? Even if you adopted my original suggestion the example below it (if fixed) would clarify what was meant.

If you adopted the briefer (and more conventional) notation, you could always add a link to a page along the lines of "how to interpret function syntax notation".

per1234, 2015-05-06:
I agree 100% about the examples using int instead of byte. I really don't see any benefit to doing that. It just teaches a bad habit and people may assume there is a good reason for it. Well now that the reference is on Github maybe something can be done about it. It would be best to change the examples included with the IDE at the same time so all the examples are consistent. I just checked the pull requests and the old Google Code page and didn't see where anyone has tried this before. Are you aware of any attempts at changing the examples to use the appropriate types in the past(other than forum discussions)? Seems like someone would have tried it already.

I hear what you're saying about the conventional notation but for me Arduino is about removing as many unnecessary **something**s that the user has to learn as possible so they can get to the fun part, even if that means the documentation is less concise. My alternative makes the information available without forcing it on the user, they can skim over it and come back later if needed. It is more efficient to just learn things the right way from the start, but if you can get beginners past the initial confusion to the having fun part quickly then they are hooked. After that they will have the determination to figure out the rest. I do think that the reference would benefit from having a link to a general overview at the top of the Variable Types section but not necessary to explain on every reference page..

nickgammon, 2015-05-06:
>  My alternative makes the information available without forcing it on the user, they can skim over it and come back later if needed.

That's fine with me. 

> Are you aware of any attempts at changing the examples to use the appropriate types in the past(other than forum discussions)?

Not personally.

sterretje, 2017-11-14:
I just like to add a vote for Nick's suggestion. When I started with Arduino about 2 years ago I found this a big omission.

darkain, 2018-05-07:
This "no data types listed in docs" just bit me in the ass again for the Nth time. The latest was confusion over what the returned type of String.chatAt() is. Does it return a single char, an Arduino String, or a two-byte C-String (character and NULL)? All three of these are plausible based on experience with other libraries in the wild on other platforms. I was able to solve this by looking at example source OUTSIDE of the Arduino documentation. That should never need to be the case for basic functions.

For a quick comparison, PHP is regarded as a very quick to pick up and learn language (the justification for not wanting datatypes listed in these docs), and yet despite PHP being a weak-typed language, they STILL list data types within their documentation to ensure zero confusion at any point with the built in libraries.

SeppPenner, 2018-06-29:
I would like to make a suggestion here, too.

[image]

I would organize the whole section from parameters to returns (See image) like the example shown below:

## Parameters

| Parameter name | Parameter type | Description                        | Optional     |
|----------------|----------------|----------------------------------|-------------------------------|
| pin            | int            |  The number of the pin on which you want to read the pulse.      |  <ul><li>[ ]  </li></ul> |
| value          | int            | type of pulse to read: either HIGH or LOW.       | <ul><li>[ ]  </li></ul> |
| timeout        | unsigned long  |  the number of microseconds to wait for the pulse to start; default is one second      | <ul><li>[x]  </li></ul> |

## Returns

Return type | Description                        |
|----------------|----------------------------------|
| unsigned long          | the length of the pulse (in microseconds) or 0 if no pulse started before the timeout ()            |

SeppPenner, 2018-07-19:
Can we poll on this? https://doodle.com/poll/bftecm7d3keh5dcx

nickgammon, 2018-07-27:
@SeppPenner Your suggestion is very good, but also the most work for the documenters to implement, and therefore the least likely to be actually done. My original suggestion of just throwing the data types into the documentation is the easiest to actually do, eg.

> **Syntax**
>
>  int digitalRead(byte pin)

Dygear, 2019-05-30:
Please, please do this! I think @SeppPenner has the most complete answer, but really _ANYTHING_ would be better than nothing. I'm running thought the documentation and this is extremely painful to figure out without it being in the reference docs.

SeppPenner, 2019-05-30:
@per1234 Can you check this, please? I do not see the error in the drone build: https://drone.arduino.cc/arduino/reference-en.

@Dygear I can only tell that it should be done already (At least for the English and the German version of the reference).

per1234, 2019-05-30:
>Can you check this, please? I do not see the error in the drone build

I don't know the cause of it either, but the documentation was published to arduino.cc and, at least in the translation repository I checked, the automated sync issue was created by ArduinoBot. Maybe it was just a temporary glitch. I seem to remember someone from Arduino saying that can happen when you had inquired about Drone build failures previously.

>I can only tell that it should be done already (At least for the English and the German version of the reference).

No, it's not done. Some of the reference pages do document parameter and return types but not all of them. Some of the existing documentation is wrong or incomplete. I would love to go through the entire content of the Arduino Language Reference, and confirm all the existing documentation as well as adding the missing documentation. I prepared the way for this work in https://github.com/arduino/reference-en/pull/559. Now we at least have a standardized format in place for documenting this information.

SeppPenner, 2019-05-31:
> Maybe it was just a temporary glitch. I seem to remember someone from Arduino saying that can happen when you had inquired about Drone build failures previously.

I already assumed that, too. You're right, I had the same issue in the German repository, too.

>  Some of the reference pages do document parameter and return types but not all of them.

Really? I haven't seen some. I need to check this again.

> Some of the existing documentation is wrong or incomplete.

Yeah, that might be. I will check this, too.

per1234, 2019-05-31:
>Really? I haven't seen some.

The very first page I checked:
https://www.arduino.cc/reference/en/language/functions/digital-io/digitalread/
Neither the parameter, nor return type is documented.

>Yeah, that might be. I will check this, too.

The tricky thing is they aren't guaranteed to be the same from one core to another. Luckily Arduino is standardizing a good deal of the core API via project chainsaw:
https://github.com/arduino/ArduinoCore-API
That isn't actually in use in any of Arduino's cores other than megaAVR, but it will be eventually so it makes sense to base the documentation on the function signatures contained in repository. arduino/ArduinoCore-API only contains the non-architecture specific core code so there are still more reference pages where the function signatures in each of the official cores would need to checked to ensure the documentation is 100% comprehensive regarding parameter and return types.

There is a good chance other issues with the documentation will be discovered at the same time if the process of checking the reference content against the function signatures in the core code is done carefully.

Dygear, 2019-06-02:
@per1234 Ironically, the digitalRead function was the exact one I was looking at.

leonty, 2019-09-02:
Hi all, 
As a software developer with many years of experience I support the topic starter first proposition. The documentation BADLY lacks of functions signatures in the way that all the c/c++/java etc developers are used to read. I realize that this kind of deviations from the documentation de-facto standards might be better for newbies, but... they are not! 
Really, how this variant of signature can be easier if it lacks required semantics and a newbie anyways must read all the article to perceive the required information? Moreover, this forces even an experienced developer to read all the article to find the return type and parameter types while he'd only need a full signature. That really wastes time of any developers not regarding their experience.
I think (as others in the discussion) that the most useful pieces of the documentation for a newbie are examples, not meaningless function signatures without types and parameters. I vote for the topic starter proposition and I'd help with bringing the signatures into the standard form.

per1234, 2019-09-03:
@leonty the decision has already been made regarding how the parameter and return types will be documented:
https://github.com/arduino/reference-en/pull/559
We will not be documenting this information via function signatures in the Arduino Language reference.

SeppPenner, 2019-09-03:
@per1234 I'm not sure yet, but can we close this as well? As far as I know, the return types and parameter types have been updated everywhere, haven't they? (At least, I remeber that you did something there and I did a lot of updates in the German repo as well).

per1234, 2019-09-03:
@SeppPenner I made the formatting of all the existing parameter and return type documentation to match the standard established by the [reference sample page](https://github.com/arduino/reference-en/blob/master/AsciiDoc_sample/Reference_Terms/AsciiDoc_Template-Single_Entity.adoc) in https://github.com/arduino/reference-en/pull/559. However, there are still many places in the documentation where parameter or return types are not documented at all and there are also places where the existing documentation is incorrect or incomplete (i.e., doesn't reflect that different types are used from one architecture to another). So although the question of how this information should be documented is settled, the need to add the documentation where missing is still very much in need. I'd like to leave it open so we can continue to use this issue for that purpose, while avoiding spending any more time discussing how to document the information.

SeppPenner, 2019-09-03:
> However, there are still many places in the documentation where parameter or return types are not documented at all and there are also places where the existing documentation is incorrect or incomplete (i.e., doesn't reflect that different types are used from one architecture to another).

> I'd like to leave it open so we can continue to use this issue for that purpose, while avoiding spending any more time discussing how to document the information.

Ah okay, I understand that now :)