arduino/reference-en #16: Documentation of Functions of StringObject is missing

utix, 2015-05-02:
Hi,
some documentations seem missing. For example this one:
http://www.arduino.cc/en/Reference/StringToCharArray

I want propose some enhancement to this doc, how can I do it ?

Thanks for your jobs
Ôrel

per1234, 2017-09-28:
In the current reference pages String (object) documentation is linked from the "Language Reference" page rather than the "Libraries" page. I don't think most users think of it as a "library" in the Arduino use of the term. However, the String (object) documentation page is formatted like the libraries. I know [Massimo Banzi has stated](https://groups.google.com/a/arduino.cc/forum/#!topic/developers/oyt_c5wePg8) that the plan is to eventually store the library documentation with the library source code but I don't know how that would work with String since it is part of the core. I suppose the documentation could be put in the core folder also but it seems inconsistent to document some of the core functions in this repository and other core functions in other places.

So my gut feeling is that this is not part of #18.

The [Serial page](https://www.arduino.cc/en/Reference/Serial) in the reference is also formatted like a library page but the [page in this repository](https://github.com/arduino/reference-en/blob/master/Language/Functions/Communication/serial.adoc) here in this reference does list all the functions (but those links happen to be broken).

ArduinoBot, 2017-10-12:
➤ Simone Majocchi commented:

Discussed with @Stefania Mellai the status of current Reference pages about "String object" and found that the content is on .CC and just needs porting.

smellai, 2017-10-17:
I found the page in the variables/data-types section: https://cdn.arduino.cc/reference/en/language/variables/data-types/stringobject/

ArduinoBot, 2017-10-20:
➤ Simone Majocchi commented:

Working on the functions related to stringObject as offline .adoc. Soon to be sent to Stefania as a Zip for proper uploading on the reference.

smellai, 2017-10-20:
we should also rename and/or move https://cdn.arduino.cc/reference/en/language/variables/data-types/string/ as array of chars is not a type

ArduinoBot, 2017-10-20:
➤ Simone Majocchi commented:

@Stefania Mellai here is the Zipped file with functions, operators and main object page. I kep the folders structure as agreed.

I hope they are OK.

smellai, 2017-10-23:
Should be fixed

closed 2017-10-23 by smellai