arduino/reference-en #18: Add Libraries Reference pages

per1234, 2015-05-07:
Please add the Libraries Reference pages: http://www.arduino.cc/en/Reference/Libraries

carlosperate, 2015-07-17:
Indeed this is still missing. Is this repository still being maintained as the source for the reference documentation? It's hard to tell.

per1234, 2017-09-28:
A post by Massimo Banzi to the Arduino Developers Mailing List:
https://groups.google.com/a/arduino.cc/forum/#!topic/developers/oyt_c5wePg8
may provide some good justification for leaving the libraries reference pages out of this repository. The idea is that the library documentation will be stored with the library source code. So that content would be found in each library's repository.

animeshsrivastava24, 2019-05-30:
@per1234  Please guide me how to move forward to solve this Issue?

per1234, 2019-05-31:
@animeshsrivastava24 there is currently no way for you to move it forward other than by expressing your support for the idea.

This can only happen if someone high up at Arduino makes it a priority and then the Arduino employees with the access and knowledge to set up the necessary infrastructure do the necessary work. I'm sure it's a big job, but the fact that we already have the system in place for the Arduino Language Reference to use as a model should be a huge help.

If the infrastructure was put in place, then there would be huge opportunities for the community to help with the project. This is one of the huge benefits of this project. With the current system, it is very inefficient to make fixes or improvement to the Arduino library reference pages (and all the non-Language Reference documentation content for that matter) for both the contributor suggesting a change and the Arduino employee implementing that change.

animeshsrivastava24, 2019-06-02:
@per1234 Thank you .Sure, I can understand. I'll definitely look forward to activity and contribution to this Issue from the Community.

per1234, 2022-04-20:
>the library documentation will be stored with the library source code

This system has now been created, the library reference content migrated to all the official library repositories, and published on arduino.cc.

So I am happy to be able to close this request as having been fulfilled in a far better manner than what I proposed 7 years ago!

closed 2022-04-20 by per1234