arduino/reference-en #19: All "See Also" links to Github pages are missing the dot before the extension

per1234, 2015-06-13:
For example the [noTone](https://github.com/arduino/reference-en/blob/master/Language/Functions/Advanced%20IO/noTone.adoc) page's **See Also** link points to:
https://github.com/arduino/reference-en/blob/master/Language/Functions/Advanced%20IO/toneadoc
Instead the link should be:
https://github.com/arduino/reference-en/blob/master/Language/Functions/Advanced%20IO/tone.adoc

per1234, 2015-06-17:
I think the solution is either:
- Change `:ext-relative: adoc` to `:ext-relative: .adoc`
- Change `* #LANGUAGE# link:tone{ext-relative}[tone()]` to `* #LANGUAGE# link:tone.{ext-relative}[tone()]`

I'm happy to make the changes and submit a pull request but I don't know which change will work correctly with how the extension is changed to .html during compilation.

00alis, 2017-07-25:
@mastrolinux could you gibe @per1234 some guidance on how to fix this?

smellai, 2017-09-28:
I think this is ok now? 
https://cdn.arduino.cc/reference/en/language/functions/advanced-io/notone/
@per1234 can you confirm?

per1234, 2017-09-28:
Looks good, thanks!

closed 2017-09-28 by per1234