arduino/reference-en #22: Status of this repo?

matthijskooijman, 2015-12-10:
It appears that this repository is the source of (a part of) the docs on arduino.cc. However, it hasn't seen any updates in a while, and I do believe doc changes have been made. Can someone clarify the status of this repo?

mastrolinux, 2015-12-14:
Hi, this is a work in progress project to move all the documentation to a github repo. It is not the official documentation yet. We will update you when there will be changes.

cousteaulecommandant, 2015-12-14:
So what's currently the official way of updating the documentation? (I recently made PR #23 because I couldn't wait; not sure if I should have used a different way instead)

matthijskooijman, 2015-12-14:
@mastrolinux, perhaps a prominent note to the readme of this repo explaining it is not yet currently used would be appropriate? That would prevent confusion among potential contributors.

per1234, 2017-07-20:
Could we get an update on this? I was told (https://github.com/arduino/reference-en/pull/36#issuecomment-304599984) that an "official announcement will be done within June this year" but still no word.

I've been doing a lot of work reviewing the arduino.cc pages lately and I'd like to know how that will be affected by this repository.

We have people fixing bugs in this repository that don't exist in the actual reference pages. We have fixes/improvements made here but not in the reference pages. We have fixes/improvements to the reference pages that are not present in this repository. How do you plan to merge this all together?

smellai, 2017-07-20:
Here is the site using this repo: https://cdn.arduino.cc/reference/en/
The site is ready but, as you said, we need to align the two sources before the switch and to find out any issue remaining (broken links or things like that). If you could open issues here about the differences you mentioned above, it would be really great! Thanks @per1234

smellai, 2017-10-19:
we announced the switch https://create.arduino.cc/projecthub/Arduino_Genuino/contribute-to-the-arduino-reference-af7c37

closed 2017-10-19 by smellai