arduino/reference-en #28: Missing pages: Mouse and Keyboard

00alis, 2016-04-19:
This page http://www.arduino.cc/en/Reference/MouseKeyboard is missing its adoc document, would it be possible to create it as well?

Thanks!

sh-aks, 2016-04-19:
@00alis: Files Added in the branch Mouse_Keyboard_Fix
Will raise a pull request, after the discussion.

Did not understand where to add the Caution which is common for both the library.
Should we put on both the pages?
[image]

00alis, 2016-04-20:
Yes please I think it is best to add it to both pages, thank you!

sh-aks, 2016-04-20:
@00alis : Added Notes and Warning Section.
Closing the issue and Making a pull request.

sh-aks, 2016-04-27:
Pending Pull request.

commit c356e143 (2016-04-27): Add Mouse.adoc, Keyboard.adoc, keyboard.Modifiers.adoc

Fix #28
closed 2016-04-27 by lorenzoromagnoli