arduino/reference-en #52: AsciiDoc_Template-Single_Entity.adoc doesn't match the reference pages in the repository

per1234, 2017-07-20:
[AsciiDoc_Template-Single_Entity.adoc](https://raw.githubusercontent.com/arduino/reference-en/master/AsciiDoc_sample/Reference_Terms/AsciiDoc_Template-Single_Entity.adoc) is missing the first section that is found in all the other reference pages in this repository. For example, https://raw.githubusercontent.com/arduino/reference-en/master/Language/Structure/Comparison%20Operators/lessThanOrEqualTo.adoc, which was created 9 days ago so I have to assume it is in the correct format, starts with:
```
---
title: "<="
title_expanded: Less Than or Equal To
categories: [ "Structure" ]
subCategories: [ "Comparison Operators" ]
---
```
but nothing of that sort is found in the example.

00alis, 2017-07-24:
@SimonePDA do you mind updating the template since you have been working on this more recently? Thanks!

00alis, 2017-07-25:
https://github.com/arduino/reference-en/blob/master/AsciiDoc_sample/Reference_Terms/AsciiDoc_Template-Single_Entity.adoc

Unfortunately adding comments to this part breaks the preview, but you are right it is more accurate now.

closed 2017-07-25 by 00alis