arduino/reference-en #54: Arduino DUE only should be Due & Zero only?

smellai, 2017-07-21:
[image]

From https://www.arduino.cc/en/Reference/HomePage:
[image]

per1234, 2017-07-21:
What about all the other Arduino SAMD Boards too?

ArduinoBot, 2017-10-11:
➤ Simone Majocchi commented:

I have checked with Arturo and the boards that have SAMD micros and therefore support those two functions are:
DUE, MKR1000, MKR Zero, MKR FOX, MKR WAN, MKR GSM.

I'm correcting the relevant pages.

ArduinoBot, 2017-10-11:
➤ Simone Majocchi commented:

I have updated and pushed the Reference home page with this new header:
Zero, DUE & MKR Family

www.arduino.cc/en/Reference/HomePage

smellai, 2017-10-11:
what about this?
[image]

SimonePDA, 2017-10-11:
@smellai where I can replicate this correction in the Github reference?

ArduinoBot, 2017-10-11:
➤ Simone Majocchi commented:

Fixed the two function pages lanalogReadResolution and analogWriteResolution with the reference to Zero, Due and MKR Family of boards.

Pushed already

closed 2017-10-11 by smellai

ArduinoBot, 2017-10-12:
➤ Simone Majocchi commented:

Fixed GIT pages as well.