arduino/reference-en #61: Correct links or use aliases

smellai, 2017-08-03:
we have some links in camelcase around the files, that's because maybe in the previous version of hugo they worked in camelcase. We should either convert them to lowercase or add an alias as in the following commit:

https://github.com/arduino/reference-en/commit/a1ebfe8c6f703e4def57f4cf9966374593cd15da

in this way both links will work:
https://cdn.arduino.cc/reference/en/language/functions/advanced-io/notone/
https://cdn.arduino.cc/reference/en/language/functions/advanced-io/noTone/

Serjlee, 2017-10-27:
Hello
While working on a scraper I found some dead links:
```
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/advanced-io/digital-io/pinMode> (referer: https://cdn.arduino.cc/reference/en/language/functions/advanced-io/shiftout/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/advanced-io/SPI> (referer: https://cdn.arduino.cc/reference/en/language/functions/advanced-io/shiftout/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/time/analog-io/analogWrite> (referer: https://cdn.arduino.cc/reference/en/language/functions/time/delay/)
(403) <GET https://cdn.arduino.cc/reference/en/language/structure/control-structures/if> (referer: https://cdn.arduino.cc/reference/en/language/structure/boolean-operators/logicalor/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/variable-scope--qualifiers/define> (referer: https://cdn.arduino.cc/reference/en/language/variables/variable-scope--qualifiers/const/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/variable-scope--qualifiers/array> (referer: https://cdn.arduino.cc/reference/en/language/variables/variable-scope--qualifiers/const/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/data-types/fpConstant> (referer: https://cdn.arduino.cc/reference/en/language/variables/data-types/float/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/data-types/int/unsignedInt%7Bext-relative%7D> (referer: https://cdn.arduino.cc/reference/en/language/variables/data-types/int/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/structure/bitwise-operators/bitshiftRight> (referer: https://cdn.arduino.cc/reference/en/language/variables/data-types/int/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/data-types/ASCIIchart> (referer: https://cdn.arduino.cc/reference/en/language/variables/data-types/char/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/data-types/println> (referer: https://cdn.arduino.cc/reference/en/language/variables/data-types/char/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/contants/constants> (referer: https://cdn.arduino.cc/reference/en/language/variables/data-types/boolean/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/conversion/data-types/word> (referer: https://cdn.arduino.cc/reference/en/language/variables/conversion/wordcast/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/data-types> (referer: https://cdn.arduino.cc/reference/en/language/variables/constants/integerconstants/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardWrite> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/mouse/mouseBegin> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/mouse/mousepress/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/mouse/mousePress> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/mouse/mouserelease/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/mouse/mouseRelease> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/mouse/mousepress/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardRelease> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardReleaseAll> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardPrintln> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardPrint> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardPress> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardEnd> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardBegin> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardModifiers> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/keyboard/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/communication/> (referer: https://cdn.arduino.cc/reference/en/language/functions/communication/stream/)
(403) <GET https://cdn.arduino.cc/reference/en/language/variables/data-types/string/functions/length> (referer: https://cdn.arduino.cc/reference/en/language/variables/data-types/stringobject/)
(403) <GET https://cdn.arduino.cc/reference/en/language/functions/usb/mouse/mouseEnd> (referer: https://cdn.arduino.cc/reference/en/language/functions/usb/mouse/mousebegin/)
```
I think automatically list somewhere the dead links on each deploy is feasible, I'll eventually look into it

SimonePDA, 2017-10-30:
Done my best with camel case related issues with links

smellai, 2017-10-31:
Hi @SimonePDA we did a clean up of redundant page and we have all those links broken because of camel case, can you help with them please?

SimonePDA, 2017-10-31:
The list is in the old one place, but it is a new one...

fixing camelcase

SimonePDA, 2017-11-03:
Fixed all the camelCases.  Still there are a couple that are related to something else.

* #LANGUAGE# link:../../arduino-due-only/analogreadresolution[analogReadResolution()]

in 
https://cdn.arduino.cc/reference/en/language/functions/analog-io/analogwrite/
and
linked by https://cdn.arduino.cc/reference/en/language/functions/analog-io/analogread/

SimonePDA, 2017-11-03:
Fixed also the link into zero-due--mkr-family

#LANGUAGE# link:../../zero-due--mkr-family/analogreadresolution[analogReadResolution()]

Serjlee, 2017-11-06:
I moved the autogenerated list of broken links to its own issue (#170), which will be completely automated soon

commit 9b5ff8b1 (2017-10-31): fixed links, Fixes #61
commit 922f17ab (2017-10-31): fixed broken links, Fixes #61
closed 2017-11-06 by Serjlee