arduino/reference-en #71: Learn more page for Structure

ArduinoBot, 2017-10-10:


ArduinoBot, 2017-10-10:
➤ David A. Mellis commented:

Maybe something short like these?

Functions. For controlling the Arduino board and performing computations.

Variables. Arduino data types and constants.

Structure. The elements of Arduino (C++) code.

ArduinoBot, 2017-10-10:
➤ Alice Pintus commented:

maybe something a bit more articulated will be helpful :)))

ArduinoBot, 2017-10-10:
➤ Alice Pintus commented:

Hi @damellis,
we are back at fixing the Reference 💃 

Could you please confirm it is best to remove these pages:
https://www.arduino.cc/en/Reference/VariableDeclaration
https://www.arduino.cc/en/Reference/FunctionDeclaration and replace them with the sentences you provided us? I don't remember exactly what we decided.

If this is the case, could you provide us with a small paragraph instead of a short sentence?
Thanks!

@gmarchiarduino @mastrolinux

ArduinoBot, 2017-10-10:
➤ David A. Mellis commented:

I'm not sure if there's more I can say that would be useful and still fit in the header of a section of the reference. That is, I think we need both a reference, that provides an index / reference of each command and tutorials that explain how to use functions, variables, etc. The tutorials part needs a lot of work, but I think something like the current VariableDeclaration and FunctionDeclaration pages are good to have.

ArduinoBot, 2017-10-10:
➤ Alice Pintus commented:

ok, so we will keep them. We are missing something similar for Structure, would you be able to write something? Thanks!

ArduinoBot, 2017-10-10:
➤ David A. Mellis commented:

I probably won't have time any time soon, but I might be able to get to it
eventually, if I'm the best / only person to write it. Maybe there's
something from Massimo's book we can use? Or another source?

ArduinoBot, 2017-10-10:
➤ Alice Pintus commented:

@SimonePDA @akash73 would it be possibile to write a documentation page on Structure similar to https://www.arduino.cc/en/Reference/VariableDeclaration
https://www.arduino.cc/en/Reference/FunctionDeclaration 

These 2 pages probably will benefit some review as well. Let us know, thanks!

ArduinoBot, 2017-10-10:
➤ Andrea Richetta commented:

@Simone Majocchi can you write a documentation page on Structure as described? thx

ArduinoBot, 2017-10-10:
➤ Simone Majocchi commented:

@Andrea Richetta what is your timeframe? It needs some careful tech writing and it needs to be done thoughtfully. I'd like to plan this activity without rushing it.

ArduinoBot, 2017-10-10:
➤ Alice Pintus commented:

We are not in a rush for this, @SimonePDA could you give an estimate of the time that you need? Thanks!

smellai, 2017-10-10:
https://github.com/bcmi-labs/reference/issues/11

closed 2017-10-10 by smellai