arduino/reference-en #134: Issues with AsciiDoc_Template-Dictionary.adoc

per1234, 2017-10-23:
[README.adoc](https://github.com/arduino/reference-en/blob/master/README.adoc) claims:
>In r[eference-en/AsciiDoc_sample/AsciiDoc_Dictionary](https://raw.githubusercontent.com/arduino/reference-en/master/AsciiDoc_sample/AsciiDoc_Dictionary/AsciiDoc_Template-Dictionary.adoc) you will find an overview on the AsciiDoc syntax. This includes Titles, Text, Links, Images, Tables, Code and various embeds (video, slideshow, audio and code). You can see it rendered [here](https://reference.arduino.cc/reference/en/asciidoc_sample/asciidoc_dictionary/asciidoc_template-dictionary/).
Several of the items in this file do not correctly render:
- "This is a block image with a URL" - the file http://store.arduino.cc/bmz_cache/b/babe9d3f8563ea2f170dd55539f06310.image.538x354.jpg does not exist.
- "This is an image saved in the attachments folder with a caption" - I get a 403 error for https://reference.arduino.cc/reference/en/asciidoc_sample/asciidoc_dictionary/asciidoc_template-dictionary/attachments/ArduinoUno_R3_Front_450px.jpg
- "This is a pretty embed of code from Gist" - The gist (http://gist.github.com/mojavelinux/5333524.js) exists but it doesn't appear on that page.
- "And this is another of a video from YouTube" - There is just a blank space where the video should be. I suspect this is caused by the use of www.youtube-nocookie.com.
- "And here some slides from Sideshare, yay!" - There is just an empty box where the slides should be.

smellai, 2017-10-24:
youtube iframe is blocked by cdn security rules, it will be fixed once in production

smellai, 2017-11-08:
youtube video still broken but not a problem for the reference, closing.

commit 72f65142 (2017-10-24): fixed asciidoc template dictionary, Fixes #134
closed 2017-11-08 by smellai