arduino/reference-en #209: Is it possible to access the old Language Reference pages?

per1234, 2017-11-09:
In the process of fixing regressions, I'm frequently wanting to look at the old reference pages so that I can port the changes here.

I did find that I could access the pre-change version of one page by using the previous URL:
https://www.arduino.cc/en/Reference/AttachInterrupt/

but with other pages that doesn't work because they redirect to the new page, for example:
https://www.arduino.cc/en/Reference/Constants
redirects to
https://www.arduino.cc/reference/en/language/variables/constants/constants

So, as a side note, I suspect that actually the lack of a redirect on https://www.arduino.cc/en/Reference/AttachInterrupt/ may be a bug.

smellai, 2017-11-09:
You are right @per1234 , we are using this redirect map but some links are wrong:
```
/en/Reference/if	https://www.arduino.cc/reference/en/language/structure/control-structure/if/;
/en/Reference/else	https://www.arduino.cc/reference/en/language/structure/control-structure/else/;
/en/Reference/for	https://www.arduino.cc/reference/en/language/structure/control-structure/for/;
/en/Reference/switchCase	https://www.arduino.cc/reference/en/language/structure/control-structure/switchcase/;
/en/Reference/while	https://www.arduino.cc/reference/en/language/structure/control-structure/while/;
/en/Reference/doWhile	https://www.arduino.cc/reference/en/language/structure/control-structure/dowhile/;
/en/Reference/break	https://www.arduino.cc/reference/en/language/structure/control-structure/break/;
/en/Reference/continue	https://www.arduino.cc/reference/en/language/structure/control-structure/continue/;
/en/Reference/return	https://www.arduino.cc/reference/en/language/structure/control-structure/return/;
/en/Reference/goto	https://www.arduino.cc/reference/en/language/structure/control-structure/goto/;
/en/Reference/SemiColon	https://www.arduino.cc/reference/en/language/structure/further-syntax/semicolon/;
/en/Reference/Braces	https://www.arduino.cc/reference/en/language/structure/further-syntax/curlybraces/;
/en/Reference/Comments	https://www.arduino.cc/reference/en/language/structure/further-syntax/blockcomment/;
/en/Reference/Define	https://www.arduino.cc/reference/en/language/structure/further-syntax/define/;
/en/Reference/Include	https://www.arduino.cc/reference/en/language/structure/further-syntax/include/;
/en/Reference/Assignment	https://www.arduino.cc/reference/en/language/structure/arithmetic-operators/assignment/;
/en/Reference/Arithmetic	https://www.arduino.cc/reference/en/language/structure/arithmetic-operators/addition/;
/en/Reference/modulo	https://www.arduino.cc/reference/en/language/structure/arithmetic-operators/modulo/;
/en/Reference/Boolean	https://www.arduino.cc/reference/en/language/structure/boolean-operators/logicaland/;
/en/Reference/pointer	https://www.arduino.cc/reference/en/language/structure/pointer-access-operators/reference/;
/en/Reference/BitwiseAnd	https://www.arduino.cc/reference/en/language/structure/bitwise-operators/bitwiseand/;
/en/Reference/BitwiseXorNot	https://www.arduino.cc/reference/en/language/structure/bitwise-operators/bitwisenot/;
/en/Reference/Bitshift	https://www.arduino.cc/reference/en/language/structure/bitwise-operators/bitshiftleft/;
/en/Reference/increment	https://www.arduino.cc/reference/en/language/structure/compound-operators/increment/;
/en/Reference/incrementCompound	https://www.arduino.cc/reference/en/language/structure/compound-operators/compoundaddition/;
/en/Reference/bitwiseCompoundAnd	https://www.arduino.cc/reference/en/language/structure/compound-operators/compoundbitwiseand/;
/en/Reference/bitwiseCompoundOr	https://www.arduino.cc/reference/en/language/structure/compound-operators/compoundbitwiseor/;
/en/Reference/Constants	https://www.arduino.cc/reference/en/language/variables/constants/constants/;
/en/Reference/IntegerConstants	https://www.arduino.cc/reference/en/language/variables/constants/integerconstants/;
/en/Reference/Fpconstants	https://www.arduino.cc/reference/en/language/variables/constants/floatingpointconstants/;
/en/Reference/void	https://www.arduino.cc/reference/en/language/variables/data-types/void/;
/en/Reference/BooleanVariables	https://www.arduino.cc/reference/en/language/variables/data-types/boolean/;
/en/Reference/char	https://www.arduino.cc/reference/en/language/variables/data-types/char/;
/en/Reference/unsignedChar	https://www.arduino.cc/reference/en/language/variables/data-types/unsignedchar/;
/en/Reference/byte	https://www.arduino.cc/reference/en/language/variables/data-types/byte/;
/en/Reference/int	https://www.arduino.cc/reference/en/language/variables/data-types/int/;
/en/Reference/unsignedInt	https://www.arduino.cc/reference/en/language/variables/data-types/unsignedint/;
/en/Reference/word	https://www.arduino.cc/reference/en/language/variables/data-types/word/;
/en/Reference/long	https://www.arduino.cc/reference/en/language/variables/data-types/long/;
/en/Reference/unsignedLong	https://www.arduino.cc/reference/en/language/variables/data-types/unsignedlong/;
/en/Reference/short	https://www.arduino.cc/reference/en/language/variables/data-types/short/;
/en/Reference/float	https://www.arduino.cc/reference/en/language/variables/data-types/float/;
/en/Reference/double	https://www.arduino.cc/reference/en/language/variables/data-types/double/;
/en/Reference/string	https://www.arduino.cc/reference/en/language/variables/data-types/string/;
/en/Reference/StringObject	https://www.arduino.cc/reference/en/language/variables/data-types/stringobject/;
/en/Reference/array	https://www.arduino.cc/reference/en/language/variables/data-types/array/;
/en/Reference/CharCast	https://www.arduino.cc/reference/en/language/variables/conversion/charcast/;
/en/Reference/ByteCast	https://www.arduino.cc/reference/en/language/variables/conversion/bytecast/;
/en/Reference/IntCast	https://www.arduino.cc/reference/en/language/variables/conversion/intcast/;
/en/Reference/WordCast	https://www.arduino.cc/reference/en/language/variables/conversion/wordcast/;
/en/Reference/LongCast	https://www.arduino.cc/reference/en/language/variables/conversion/longcast/;
/en/Reference/FloatCast	https://www.arduino.cc/reference/en/language/variables/conversion/floatcast/;
[... cut ...]
/en/Reference/StringGE	https://www.arduino.cc/reference/en/language/variables/data-types/string/operators/greaterthanorequalto;
/en/Reference/StringLT	https://www.arduino.cc/reference/en/language/variables/data-types/string/operators/lessthan;
/en/Reference/StringLE	https://www.arduino.cc/reference/en/language/variables/data-types/string/operators/lessthanorequalto;
/en/Reference/StringDF	https://www.arduino.cc/reference/en/language/variables/data-types/string/operators/differentfrom;
```
I'm going to remove the redirects to help you and the community with the review. Thank you!

per1234, 2017-11-09:
Thanks, that made the work immensely easier.

I believe I've now mined the closed issues labeled `Component: Documentation` in the arduino/Arduino issue tracker fairly thoroughly. There may still be a few left in the really old issues that were not fixed until recently but the yield was getting pretty low so I stopped at June, 2014.

I've submitted PRs for all of the major regressions I found through that process except this one:
https://github.com/arduino/Arduino/issues/2496#issuecomment-66872901
(the comment by Paul Stoffregen)
I don't have the energy to tackle that right now so if someone else wants to take over you're welcome to it, otherwise I'll try to get to it soon.

I believe the rest of the work on my to-do list can be done without access to the previous pages but I suspect there are other regressions that were not exposed by looking through the closed issues. It's frustrating to see improvements to the documentation make over the last couple of years gone. I am really excited about this new reference system but it's unfortunate that it's all based on significantly outdated content. I'm guessing you'll want to get those redirects turned back on ASAP but it would be really nice it there was still some way for people working on the documentation content to still have access to the previous pages.

smellai, 2017-11-14:
I restored the redirects and opened a new issue for the thing you reported. Thanks

closed 2017-11-14 by smellai