arduino/reference-en #228: Suggestion to add size_t reference page

per1234, 2017-11-12:
*Moved from https://github.com/arduino/Arduino/issues/6921 by @sterretje

As title says

Add a data type size_t to the data type section; maybe add a hint about sizeof to determine what the size of size_t is so people can determine for themselves.

per1234, 2017-11-12:
I agree that it should be documented. Although I don't think size_t is a type that will be commonly used by beginners in their code, some Arduino functions return size_t, which should be documented, and that will naturally lead users to look for a size_t page in the Data Types section of the Language Reference.

q2dg, 2018-07-20:
Well, datatypes documentation is a lost battle...see https://github.com/arduino/Arduino/issues/4525