arduino/reference-en #234: Grammar corrections in ethernet reference

sterretje, 2017-11-13:
https://www.arduino.cc/en/Reference/Ethernet

I picked up the following. Note that I'm not native English speaking, so might have some things wrong.

> The libraries are allow an Arduino board to connect to the internet.
'are' to be removed

> The libraries support up to four concurrent connection (incoming or outgoing or a combination). 
'connection' to be changed to 'connection**s**'

per1234, 2017-11-13:
The suggestions are correct. Thanks!

Unfortunately this repository is only for the [Language Reference](https://www.arduino.cc/reference/en/) pages and should only be used to report issues with those pages. I believe the best place to submit issues with library documentation would be to that library's repository. You can find all the official Arduino libraries here:
https://github.com/arduino-libraries

The only libraries you will not find at the above link are the libraries bundled with hardware packages (e.g. EEPROM, SPI, Wire), which should be reported in that package's repository, for example Arduino AVR Boards is here:
https://github.com/arduino/ArduinoCore-avr

The documentation for the Keyboard and Mouse libraries somehow were mistakenly put in this repository and so we can edit or report issues with that content here. I'm sure that was the cause of the confusion.

sterretje, 2017-11-14:
Thanks for the reply.

grumpf ;-)

What do you suggest? Close this one and open a new one in https://github.com/arduino-libraries/Ethernet/issues referring to this one.

I don't want to create a bigger 'mess' than I've already done ;-)

per1234, 2017-11-14:
No worries. Your contributions are much appreciated! There are so many repositories that it gets confusing but there are also lots of issues so it's important that we keep them well organized otherwise they may get lost in the pile for years.

If you wouldn't mind closing this one and opening a new one in the Ethernet repository, that would be great. Unfortunately we can't submit pull requests for the changes to the library documentation content so submitting an issue is the best way to proceed.

sterretje, 2017-11-14:
Submitted a new one: https://github.com/arduino-libraries/Ethernet/issues/53

closed 2017-11-14 by sterretje