arduino/reference-en #236: https://www.arduino.cc/en/Reference/AttachInterrupt/ not redirected

smellai, 2017-11-14:
Both https://www.arduino.cc/en/Reference/max and https://www.arduino.cc/en/Reference/Max are correctly redirected to the new reference page, while AttachInterrupt not

smellai, 2017-11-15:
This page is still available https://www.arduino.cc/en/Reference/KeyboardPress

endorama, 2019-07-10:
Added redirects:
- https://www.arduino.cc/en/Reference/KeyboardPress => https://www.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardpress/
- https://www.arduino.cc/en/Reference/max/ => https://www.arduino.cc/reference/en/language/functions/math/max/

(expect cache to last few hours)

endorama, 2019-07-12:
Redirects are working, closing.

per1234, 2019-07-13:
@endorama the https://www.arduino.cc/en/Reference/max/ URL was only an example of the problem. The issue of old reference URLs not redirecting if they have a trailing slash occurs for **all** old reference URLs (except for https://www.arduino.cc/en/Reference/max/ now).

For example:
https://www.arduino.cc/en/Reference/min/
still doesn't redirect even though:
https://www.arduino.cc/en/Reference/min
does redirect.

endorama, 2019-07-15:
@per1234 please have a look at it now, I should have been able to provide a fix

closed 2019-07-18 by per1234