arduino/reference-en #270: Strange first character at a file.

sebuls, 2017-11-25:
First character at
https://github.com/arduino/reference-en/edit/master/Language/Variables/Data%20Types/String/Operators/append.adoc
is strange.

What is it ?

SeppPenner, 2018-06-27:
Looks better now: https://github.com/SeppPenner/reference-en/blob/ab9210139534791e9a06a4de7f4c62941ad5d428/Language/Variables/Data%20Types/String/Operators/append.adoc

PR is there: https://github.com/arduino/reference-en/pull/384

SeppPenner, 2018-06-27:
Is fixed in the German repository ;) https://github.com/arduino/reference-de/pull/21

However, someone should fix this here using my pull request.

SeppPenner, 2018-06-27:
PR is there for spanish repo, too. https://github.com/arduino/reference-es/pull/112

SeppPenner, 2018-06-27:
Fixed in French: https://github.com/arduino/reference-fr/pull/13

SeppPenner, 2018-06-27:
Fixed in italian repo: https://github.com/arduino/reference-it/pull/112

SeppPenner, 2018-07-09:
I think, this issue can be closed. Seperate PRs are already present in any other affected language repository.

SeppPenner, 2018-07-19:
@mastrolinux Can you close this, please. Thank you ;)

closed 2018-07-19 by sebuls