arduino/reference-en #288: Not all translation repositories get automatically generated issues

per1234, 2017-12-17:
The reference-es, reference-it, reference-jp, and reference-pt repositories get automated issue reports when this repository is updated but reference-de, reference-fr, and reference-ko do not. The first two may be intentional since they have not had any translation activity yet but surely at the least reference-ko should also receive the automated issue reports.

mastrolinux, 2018-01-16:
reference-ko should be fixed few days ago. Are you still having the same issue?

smellai, 2018-01-16:
yes yesterday an issue was open in other repos but not -ko

mastrolinux, 2018-01-17:
should be fixed now. It was missing an "automated" label so the integration was failing during the issue creation. Please test again and let me know

closed 2018-01-19 by smellai