arduino/reference-en #296: Misleading description of function parameters

nickgammon, 2017-12-26:
On this page: https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/

The function parameters are described as:

> Parameters
>
>**pin**: the pin to write to. Allowed data types: int.
>**value**: the duty cycle: between 0 (always off) and 255 (always on). Allowed data types: int

This simply isn't true. The function prototype from Arduino.h is:

```void analogWrite(uint8_t, int);```

For one thing, the first parameter (pin) is not **int**, it is **uint8_t**

Second, what does "allowed" mean? Here is some example code:

```
void setup() {
  byte foo = 5;
  analogWrite (foo, 666);
}

void loop() { }
```

That compiled without errors or warnings, even with warnings set to "all". If I set "foo" to be "long" type it still compiles OK. So really, a lot of types are "allowed". It is somewhat misleading to suggest that the pin number should be an **int**. For one thing, that implies negative pin numbers which surely is nonsense. For another, it implies that pin numbers can be greater then 255.

---

I haven't checked the other functions but starting with this one it seems that someone should go through them all and make them reflect what is in Arduino.h. Also, change the wording of "allowed" to something else.

SimonePDA, 2018-01-02:
Dear Nick,
The Arduino programming language is a modified version of C and we are aware that the data types used in our examples are less specific than what C would allow. This is a choice that was made by the Founders and that we try to keep throughout all our examples. The Arduino.h is a file that users don't usually look at and we really want to keep Arduino simple, especially in the examples. You are free to use standard ANSI C data types in your sketches, they will work, but that is beyond the scope of Arduino as a learning platform.

closed 2018-01-02 by SimonePDA

nickgammon, 2018-01-03:
I don't have any objection to keeping things simple, and I acknowledge that it is a learning platform. However **accuracy** in programming is also something worth teaching. The **uint8_t** type is also widely known in the Arduino examples as **byte**. Would it really hurt to be accurate? Just say:

```pin: the pin to write to. Data type: byte.```

Would you be OK with "allowed data types" being **Int** (note capitalization error) or **integer** (which isn't a type). If the type is uint8_t (or byte) why not say so?

---

When you write:

> Allowed data types: int.

There is a strong implication that other types are not allowed. Why not say:

```Data type: int.```

(If the data type is indeed int)

And if we are trying to teach people to program, and if the data type is float, you wouldn't say "allowed data types: int" would you?