arduino/reference-en #301: Breadcrumb parts not working at link

teo1978, 2018-01-14:


smellai, 2018-04-18:
Hi @teo1978 , you are totally right, we already have an issue in the proper repo, thanks
https://github.com/bcmi-labs/reference/issues/73

closed 2018-04-18 by smellai

teo1978, 2018-04-18:
Oh, I see. Do you have a working link to that?
Or is it a private repo or something?

smellai, 2018-04-18:
It's private, I'll write something here too when fixed

FalconFour, 2020-06-08:
Can I +1 this? This is quite a frustrating problem when trying to browse the reference pages. A lot of reference pages have this issue, or no breadcrumb trail at all...

Examples:
https://www.arduino.cc/reference/en/language/structure/bitwise-operators/bitwisexor/
https://www.arduino.cc/reference/en/language/functions/communication/serial/println/

Don't understand why this issue is "closed" as it's been broken this whole time...

per1234, 2020-06-08:
@FalconFour I'm sorry to hear this limitation of the Arduino Language Reference is an inconvenience to you.

The reason the issue is closed is because this repository is only for the language reference content. There is no way for this issue to be fixed by changes to this repository. So, even though the issue is valid, it's off topic for this repository. There is a separate closed source repository that contains the code for generating the language reference from the content in this repository. It is in that repository that the changes could be made to fix the issue, so it is appropriate for the issue to be located in that repository, which it is.