arduino/reference-en #312: Deadlinks in documentation

gaurav-nelson, 2018-02-26:
Hello,

I am the autor of [asciidoc-link-check](https://github.com/gaurav-nelson/asciidoc-link-check), a tool to check if the hyperlinks used in an asciidoc file are live (or dead).

I ran it on your docs repository and found following dead links:

```
FILE: ./AsciiDoc_sample/AsciiDoc_Dictionary/AsciiDoc_Template-Dictionary.adoc
[✖] www.slideshare.net/damellis/introduction-to-arduino-4430242
[✖] www.slideshare.net/damellis

FILE: ./Language/Variables/Utilities/PROGMEM.adoc
[✖] http://www.nongnu.org/avr-libc/user-manual/group
```

You can view the detailed report at [accompanying Gist](https://gist.github.com/gaurav-nelson/beda05cf2054b499467598109ce89e5b#file-report-txt)

I think most of these issues could be caught prior to publishing content in the repository. We use a post-commit hook to check for all dead hyperlinks. Maybe this could be added to your project's CONTRIBUTING guide.

To use the post-commit hook, authors can create a `post-commit.sh` file, inside the `.git` folder of the docs repository.

```bash
#!/bin/bash 
echo -e "\e[32mPOST COMMIT HYPERLINKS CHECK\e[0m" 
COMMIT_HASH="$(git rev-parse @~)" #get the previous commit hash 

#run the loop for every modified or added file
for i in $(git diff --name-only "${COMMIT_HASH}") ; do

    fileList[$N]="$i"
    echo "$(asciidoc-link-check $i -p)" #check all hyperlinks in this file
    echo $'\n'
    let "N= $N + 1" 
done 
echo -e "\e[32mDONE!\e[0m"
```
Now after every commit, this script checks hyperlinks in all modified or newly added files.

smellai, 2018-02-26:
Hi @gaurav-nelson, thank you this seems really helpful and we can try to introduce it

closed 2018-08-09 by gaurav-nelson

mastrolinux, 2018-08-10:
I am not sure about this, it cannot detect the FILE: ./Language/Structure/Sketch/setup.adoc
links because they are generated only after the build so the tools only spotted 2 broken links. It is still good but not better than having a check after the build is complete. I had a script a while ago which did the same checks by using wget on the generated html, that seemed more complete. It was based on https://www.digitalocean.com/community/tutorials/how-to-find-broken-links-on-your-website-using-wget-on-debian-7