arduino/reference-en #349: Missing link to www.arduino.cc/en/Reference/UsingAVR

harlock974, 2018-04-12:
Hello
Could you bring back this link in the Reference or Librairies page ?
Regards

animeshsrivastava24, 2019-07-24:
@per1234 Please can you guide me on how to fix this issue.

per1234, 2019-07-24:
I don't remember there ever having been a link to that page in the Arduino Language Reference. However, there did used to be a link directly to the [avr-libc documentation](https://www.nongnu.org/avr-libc/user-manual/pages.html) on the Language Reference home page, which was removed during the switch to the new Language Reference system.

Although Arduino started as only supporting AVR-based microcontrollers, it is now for all architectures. So I don't know that dedicating space on the home page to an AVR-only link makes sense any more. However, AVR is still likely the most popular architecture for Arduino users so I would like for it to be easy for them to learn that there is all this additional functionality available.

I think it really is part of the https://github.com/arduino/reference-en/issues/623 discussion. If the first proposal I made in that issue was chosen, there would be a dedicated page added to the Language Reference that explains the difference between the Arduino programming language and C++ and has links to external resources for the information not provided in the Arduino Language Reference. It would make good sense to have a link to the avr-libc documentation on that page (as well as the equivalent for any other officially supported architectures).

So I don't see a way to move forward on this right now, but I'd be interested to hear any proposals.