arduino/reference-en #385: Adding the CC license image to the readme

SeppPenner, 2018-06-29:
I think it would be quite cool to add the CC license image to the readme file of the repository.

See https://creativecommons.org/choose/results-one?license_code=by-sa&amp;jurisdiction=&amp;version=3.0&amp;lang=en

[image]<br />This work is licensed under a <a rel="license" href="http://creativecommons.org/licenses/by-sa/3.0/">Creative Commons Attribution-ShareAlike 3.0 Unported License</a>.

SeppPenner, 2018-07-13:
```
image:https://i.creativecommons.org/l/by-sa/3.0/88x31.png[Creative Commons License, title="Creative Commons License"] This work is licensed under a link:https://creativecommons.org/licenses/by-sa/3.0/deed.en[Creative Commons Attribution-ShareAlike 3.0 Unported License].
```

from [https://creativecommons.org/choose/results-one?license_code=by-sa&amp;jurisdiction=&amp;version=3.0&amp;lang=en](https://creativecommons.org/choose/results-one?license_code=by-sa&amp;jurisdiction=&amp;version=3.0&amp;lang=en)

smellai, 2018-07-13:
you can provide a PR yourself :)

SeppPenner, 2018-07-13:
Hi @smellai I have done that for the German repo already. I can provide one for this repo, too.

SeppPenner, 2018-07-13:
There it is: https://github.com/arduino/reference-en/pull/388

When this is merged, we can check if the German repo gets the change notifiation, too.

closed 2018-07-13 by smellai