arduino/reference-en #462: Question to the community: Does it make sense to reference (link) all occurences of functions, constants and so on?

SeppPenner, 2018-10-07:
For example: https://www.arduino.cc/reference/en/language/functions/digital-io/digitalread/ does only highlight the constants `HIGH` and `LOW`.

In my opinion it would make more sense to add links to all occurences of functions, constants and so on.

What do you guys think?

per1234, 2018-10-08:
I like lots of links. It makes it super easy for a beginner to access the relevant reference page without any searching.

Note that the actual reference pages render links that use code tags differently than GitHub does. Consider this markup:
https://github.com/arduino/reference-en/blob/f01981d276dd4d035febd9c65e8c942887876a07/Language/Structure/Further%20Syntax/include.adoc#L27
This is how it looks on the reference page:
https://www.arduino.cc/reference/en/language/structure/further-syntax/include/
[image]
The code tags cause the link to not be colored teal, as you would expect. I don't know if this is intentional or a bug in the Arduino website. Ordinarily I find this quite annoying because you never know it's a link until you hover the mouse pointer over that text. For this reason, I prefer to not use code tags on links, so that it will be colored teal. If code tags are used on subsequent instances of linkified reference terms, this will avoid the "sea of teal" visual impact of redundant linkification, which might be an argument against.

SeppPenner, 2018-10-09:
> I like lots of links. It makes it super easy for a beginner to access the relevant reference page without any searching.

That was my idea, too.

> Note that the actual reference pages render links that use code tags differently than GitHub does. Consider this markup:

I would also recommend to add everthing with the `link:..` code tags.

The reason why I have started this, is because @robsoncouto declined my pull request: https://github.com/arduino/reference-en/pull/435 with the comment 

> Hi @SeppPenner, just a reference to the #define page in the text is enough.
> Besides, there is another reference in the See also section.

In my opinion, it's more easy for beginners if everthing is linked.

SeppPenner, 2019-07-15:
I agree with @per1234, that we should use this kind of style for the links: https://github.com/arduino/reference-en/issues/546#issuecomment-509971175