arduino/reference-en #470: Error in description of HIGH constant

auv-ee, 2018-12-04:
In:
Reference > Language > Variables > Constants > Constants
the description of the HIGH constant, in the second paragraph beginning:
"A pin may also be configured as an INPUT with pinMode(), and subsequently made HIGH with digitalWrite()."
the text should read "... INPUT_PULLUP with pinMode()..."

per1234, 2018-12-04:
That statement is actually correct. The following code:
```cpp
pinMode(pin, INPUT);  // pins are in INPUT mode by default so this is only necessary if they have previously been set to OUTPUT mode
digitalWrite(pin, HIGH);
```
Does the same thing as this code:
```cpp
pinMode(pin, INPUT_PULLUP);
```

I do think the wording of that section could be improved. It seems quite vague the way it's written. Actual code examples like what I did above would be much easier to understand (and more concise) than text describing code.

`pinMode(pin, INPUT_PULLUP);` should be encouraged since it makes the purpose of the code much more clear. It would make sense to state that in the reference page.

robsoncouto, 2018-12-04:
Hi @auv-ee 
Thanks for the suggestion.
Do you mean this paragraph?

> A  pin may also be configured as an INPUT with pinMode(), and subsequently made HIGH with digitalWrite(). This will enable the internal 20K pullup resistors, which will pull up the input pin to a HIGH reading unless it is pulled LOW by external circuitry. This is how INPUT_PULLUP works and is described below in more detail.

You can enable the pullup resistors as the writer says. It is one way of doing it, this is why they later say "This is how INPUT_PULLUP works". INPUT_PULLUP is only explained later in the text, there is no need to discuss it there, as the writer is explaining how to use the HIGH constant.

Does it make sense?

auv-ee, 2018-12-04:
Sorry for the noise.  It didn't make sense after reading it several times, but I guess I was looking at it wrong.  50yrs in EE, but 0 for Arduino.  Over and out.

robsoncouto, 2018-12-04:
No problem at all :)
As @per1234 said, the wording of the section is quite vague.

commit be0447b9 (2018-12-21): Improved description of HIGH constant.

This fixes #470
closed 2018-12-21 by robsoncouto