arduino/reference-en #473: Serial.write(buf,len) missing description of len parameter

sterretje, 2019-01-03:
There is no description of the 'len' parameter
> Syntax
Serial.write(val)
Serial.write(str)
**Serial.write(buf, len)**

Arduino Mega also supports:
Serial1, Serial2, Serial3 (in place of Serial)

Parameters
val: a value to send as a single byte
str: a string to send as a series of bytes
buf: an array to send as a series of bytes

Returns
size_t
write() will return the number of bytes written, though reading that number is optional

robsoncouto, 2019-01-03:
Yes. That is correct.
The Parameter section  is missing:
> len: the length of the array

Thanks for pointing it out :)

sterretje, 2019-01-04:
> len: the length of the array

But that's not quite correct; you can have a 100 byte array but only want to send 10 bytes.

robsoncouto, 2019-01-04:
> But that's not quite correct; you can have a 100 byte array but only want to send 10 bytes.

Oops, that is right. My bad.

How about the following?

> len: the number of bytes to be sent from the array

sterretje, 2019-01-05:
I think beginners can live with that ;)

robsoncouto, 2019-01-05:
Alright then. 
Thanks again :)

commit 2b8af425 (2019-01-03): Added missing parameter description to Serial.write()

This fixes #473
closed 2019-01-05 by robsoncouto