arduino/reference-en #474: Outdated version of the Language Reference homepage still available online

per1234, 2019-01-17:
`https://www.arduino.cc/en/Reference/Extended` appears as the second result in a Google search for `arduino language reference` and the third for `arduino language`. This page redirects to `https://www.arduino.cc/en/Reference/HomePage?from=Reference.Extended`, which is an outdated version of https://www.arduino.cc/reference/en/.

Is it possible to redirect `https://www.arduino.cc/en/Reference/Extended` to https://www.arduino.cc/reference?

endorama, 2019-07-10:
It seems that https://www.arduino.cc/en/Reference/Extended now correctly redirect to https://www.arduino.cc/reference/en/ already.

I added the redirect from https://www.arduino.cc/en/Reference/HomePage2 to https://www.arduino.cc/reference/en/

(expect cache to last for some hours)

smellai, 2019-07-10:
Let's remove the old page from pmWiki as soon as the redirect works

endorama, 2019-07-12:
Redirects are working, closing.

@smellai could you take care of removing the old page please?

closed 2019-07-12 by endorama