arduino/reference-en #481: Many old reference pages still don't redirect

per1234, 2019-01-23:
Many old reference pages still don't redirect (not necessarily a comprehensive list:
- `https://www.arduino.cc/en/Serial/Available`
- `https://www.arduino.cc/en/Serial/Begin`
- `https://www.arduino.cc/en/Serial/End`
- `https://www.arduino.cc/en/Serial/Read`
- `https://www.arduino.cc/en/Serial/Peek`
- `https://www.arduino.cc/en/Serial/Flush`
- `https://www.arduino.cc/en/Serial/Print`
- `https://www.arduino.cc/en/Serial/Println`
- `https://www.arduino.cc/en/Serial/Write`
- `https://www.arduino.cc/en/Reference/SerialEvent`
- `https://www.arduino.cc/en/Reference/StreamAvailable`
- `https://www.arduino.cc/en/Reference/StreamRead`
- `https://www.arduino.cc/en/Reference/StreamFlush`
- `https://www.arduino.cc/en/Reference/StreamFind`
- `https://www.arduino.cc/en/Reference/StreamFindUntil`
- `https://www.arduino.cc/en/Reference/StreamPeek`
- `https://www.arduino.cc/en/Reference/StreamReadBytes`
- `https://www.arduino.cc/en/Reference/StreamReadBytesUntil`
- `https://www.arduino.cc/en/Reference/StreamRead`
- `https://www.arduino.cc/en/Reference/StreamReadString`
- `https://www.arduino.cc/en/Reference/StreamReadStringUntil`
- `https://www.arduino.cc/en/Reference/StreamParseInt`
- `https://www.arduino.cc/en/Reference/StreamParseFloat`
- `https://www.arduino.cc/en/Reference/StreamSetTimeout`
- `https://www.arduino.cc/en/Reference/KeyboardBegin`
- `https://www.arduino.cc/en/Reference/KeyboardEnd`
- `https://www.arduino.cc/en/Reference/KeyboardPress`
- `https://www.arduino.cc/en/Reference/KeyboardPrint`
- `https://www.arduino.cc/en/Reference/KeyboardPrintln`
- `https://www.arduino.cc/en/Reference/KeyboardRelease`
- `https://www.arduino.cc/en/Reference/KeyboardReleaseAll`
- `https://www.arduino.cc/en/Reference/KeyboardWrite`
- `https://www.arduino.cc/en/Reference/MouseBegin`
- `https://www.arduino.cc/en/Reference/MouseClick`
- `https://www.arduino.cc/en/Reference/MouseEnd`
- `https://www.arduino.cc/en/Reference/MouseMove`
- `https://www.arduino.cc/en/Reference/MousePress`
- `https://www.arduino.cc/en/Reference/MouseRelease`
- `https://www.arduino.cc/en/Reference/MouseIsPressed`

I notice these are all from the library-style reference pages. In the case of Keyboard and Mouse, they really have no business being in the Language Reference any more since this code was broken out into libraries in Arduino IDE 1.6.6. On the other hand, having the Mouse and Keyboard content in the language reference makes it so we can actually get edits done on it so I won't complain too loudly.

Related issues:
- https://github.com/arduino/reference-en/issues/209
- https://github.com/arduino/reference-en/issues/236 (but note that none of the above end with `/` so this is not a duplicate.
- https://github.com/arduino/reference-en/issues/474

endorama, 2019-07-10:
Added all required redirects!

Please note that arduino.cc/en/Reference/KeyboardPress has been reported in #236 also (not particularly interesting but webserver supports only 1 redirect per URL so be careful)

(expect cache to last few hours)

endorama, 2019-07-12:
Redirects are working, closing.

per1234, 2019-07-13:
@endorama I found a couple of issues so I'm re-opening this:
- https://www.arduino.cc/en/Reference/SerialEvent: doesn't redirect
- https://www.arduino.cc/en/Reference/StreamReadStringUntil: I provided the wrong redirect URL. It should have been https://www.arduino.cc/reference/en/language/functions/communication/stream/streamreadstringuntil/. Sorry about that!

Everything else is working perfectly. Thanks!

endorama, 2019-07-15:
I fixed both. Please wait for cache to expire.

closed 2019-07-18 by per1234