arduino/reference-en #487: "See also" links are not automatically generated for reference pages in sub-subcategories

per1234, 2019-01-24:
Normally, links are automatically added to the "See also" section for all other pages in the same subcategory. However, when the pages are deeper than one level down, this does not happen. The result is that the "See also" sections for the Serial, Stream, Keyboard, and Mouse function reference pages only have links to the other pages in their subcategory when they have been manually added.

Reference:
https://github.com/arduino/reference-en/blob/87a3a6763645125f00ad649d803a6d84f0a8e549/AsciiDoc_sample/Reference_Terms/AsciiDoc_Template-Single_Entity.adoc#L120

kengdahl, 2020-08-27:
It is done for:

- **Stream**
- **Keyboard**
- **Mouse**

But still not for:
- **Stream**

per1234, 2020-08-27:
>It is done for:
- Stream

...

>But still not for:
- Stream

Huh? Did you make a typo?

@kengdahl this feature has been completely broken ever since the reworking of the Language Reference website generation system: https://github.com/bcmi-labs/reference/issues/119 (FYI to normal users: this is a private repo, so you getting a 404 is expected).

Any "See Also" links you see now on the website are there because they were manually added to the .adoc files. Manual "See also" links were added for either of two reasons:
- Relevant content outside the category of the page (and thus not automatically generated)
- Relevant content inside the category of the page for the few pages where the automatic "See also" link generation system was [mysteriously broken](https://github.com/bcmi-labs/reference/issues/119) (back in the good old days when it worked at all)

kengdahl, 2020-08-28:
Yes. sorry. Typo. **Stream** is missing **See also**.

I'll see if I can add the **See also** manually.