arduino/reference-en #535: Possible inconsistency in reference example re: external links

per1234, 2019-02-06:
In the "See also" section, the reference example has this to say about external links:
https://github.com/arduino/reference-en/blob/afde77c8537eb03d60ef066011bdbd425722c90a/AsciiDoc_sample/Reference_Terms/AsciiDoc_Template-Single_Entity.adoc#L127-L128

However, in the Description, there is an external link which does not use the syntax to open in a new window/tab:
https://github.com/arduino/reference-en/blob/afde77c8537eb03d60ef066011bdbd425722c90a/AsciiDoc_sample/Reference_Terms/AsciiDoc_Template-Single_Entity.adoc#L29

Although I don't have strong feelings either way about opening links in new windows/tabs, it seems a bit inconsistent to handle external links one way in a specific section, and a different way in the rest of the document. Is this intentional policy, or simply an oversight?

If the latter, I'm happy to submit a PR to correct the reference example as well as the significant number of reference pages which don't open non-See also external links in a new window/tab.

animeshsrivastava24, 2019-07-20:
@per1234  I clearly understand this Issue. Please guide me on how to fix this one. 

> Although I don't have strong feelings either way about opening links in new windows/tabs, it seems a bit inconsistent to handle external links one way in a specific section, and a different way in the rest of the document. Is this intentional policy, or simply an oversight?

Yes, I definitely agree with your views.

per1234, 2019-07-21:
Since this involves possibly changing the official style policy for the language reference, I think I need someone else from Arduino to provide a decision on how to proceed. Until then, there is nothing you can do to fix it.