arduino/reference-en #537: Incorrect redirects for old setup and loop pages

per1234, 2019-02-06:
https://www.arduino.cc/en/Reference/Setup
redirects to:
https://www.arduino.cc/reference/en/language/functions/main/setup/
but the correct URL is:
https://www.arduino.cc/reference/en/language/structure/sketch/setup/

---
https://www.arduino.cc/en/Reference/Loop
redirects to:
https://www.arduino.cc/reference/en/language/functions/main/loop/
but the correct URL is:
https://www.arduino.cc/reference/en/language/structure/sketch/loop/

---
This is not necessarily a comprehensive list. It's just the two instances I stumbled across on the "See also" section of this page:
https://www.arduino.cc/en/Tutorial/Sketch

There are previous issues related to redirects from the old reference URLs, but those have a difference cause than this one:

- https://github.com/arduino/reference-en/issues/236
- https://github.com/arduino/reference-en/issues/474
- https://github.com/arduino/reference-en/issues/481

endorama, 2019-07-10:
updated redirects to `setup` and `loop`.

`pinMode` was also wrong, fixed.

(expect cache to last few hours)

endorama, 2019-07-12:
Redirects are working, closing.

closed 2019-07-12 by endorama