arduino/reference-en #545: Unnecessary anchor tags in markup

per1234, 2019-02-18:
The markdown for each reference page creates a few anchor tags:
```asciidoc
[#overview]
[#howtouse]
[#see_also]
```
However, it's not clear to me that these are needed. The reason is that anchors are automatically generated for each heading in the page. This means that https://www.arduino.cc/reference/en/language/functions/digital-io/digitalread/#_example_code works just as well as https://www.arduino.cc/reference/en/language/functions/digital-io/digitalread/#howtouse. For this reason, it seems like the manually added anchors only clutter up the reference source, making the content less friendly to casual contributors. Perhaps there is some purpose to these I'm not aware of? If not, I propose that I remove them.

SeppPenner, 2019-06-13:
@per1234 I haven't seen the purpose of these either. I first thought that they are needed to fill e.g. the see also section with links from pages on the same folder level (But that's not true).

animeshsrivastava24, 2019-06-26:
Yes, I too have a similar doubt regarding them. In initial process of grammar check and typo correction and it's preview both after modifications and on arduino.cc I didn't found their use or visibility ( I thought I missed something).
Also while few translations, I landed upon them again and checked whether they made an importance if left untreated (I didn't found a problem).
While searching for their importance, I landed on this website

https://daveabrock.com/2018/03/03/using-anchor-links-in-markdown/

The information mentioned on the website is as follows:

> I wasn't sure how to use anchor links in Markdown. These come in handy when you have a long post and want to link to different sections of a document for easy navigation.

> The nice thing about Markdown is that it plays so well with straight HTML - so I was pleased to get it working on the first try.

> First, add an anchor as regular HTML in your Markdown element. Here, it is right at a heading.

`### <a id="MyHeading"></a>My Heading ###`

> Now, I can link it using a standard Markdown link.

`Where is my [heading](#MyHeading)?`

@per1234 @SeppPenner Please guide, does the information reflects correct use of anchor tags or I'm missing something?

SeppPenner, 2019-06-26:
> https://daveabrock.com/2018/03/03/using-anchor-links-in-markdown/

This won't apply because it's not markdown but ASCIIDoc?

animeshsrivastava24, 2019-06-26:
Yes @SeppPenner I totally agree with you. And also, I'm confused with anchor tags use.
I thought to show their relevance and general use. I guess there might be a possibility that previous comment may lead the discussion to think it was for adoc , I'll surely modify it (thank a lot for this part).