arduino/reference-en #549: Title and so on is shown incorrectly in string.remove + The example code is not opened in a new tab (Missing ^ sign)

SeppPenner, 2019-02-18:
@per1234: I think you made some mistakes in this file: https://github.com/arduino/reference-en/blob/master/Language/Variables/Data%20Types/String/Functions/remove.adoc

[image]

And the example code is not opened in a new tab (Missing `^` sign)

Original PR is https://github.com/arduino/reference-en/commit/b3e12fbffc49ada771e45e965037d98e2f301657

SeppPenner, 2019-02-18:
I've created a PR: https://github.com/arduino/reference-en/pull/550

per1234, 2019-02-18:
That was none of my doing. The file was encoded as UTF-8 BOM from the first moment it was added to the repository, long before I ever touched it.

Thanks so much for spotting the problem and fixing it! This UTF-8 BOM encoding has been a consistent problem for the project. I'll do a comprehensive scan of the entire repository today so this can be completely resolved.

closed 2019-02-18 by per1234

SeppPenner, 2019-02-22:
@per1234 Sorry for that...

I know, I already saw a lot of these strange characters and always created some issues for that.