arduino/reference-en #575: attachInterrupt example code problem

per1234, 2019-03-27:
*Moved from https://github.com/arduino/Arduino/issues/8711 by @Omar-alSuntawi*

This code does not work correctly !

```
const byte ledPin = 13;
const byte interruptPin = 2;
volatile byte state = LOW;

void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
}

void loop() {
  digitalWrite(ledPin, state);
}

void blink() {
  state = !state;
}
```

which is on the Arduino Reference [https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/](url)

It seems to accidentally read random values.

per1234, 2019-03-27:
I suspect that you are triggering the interrupt with something that has contact bounce, resulting in multiple interrupt triggers in very rapid succession and thus a random LED state for each press of the button. This could be be resolved by adding some debouncing code to the example, but I'm not sure whether this would be a good thing. The example is intended to be a simple demonstration of how to use `attachInterrupt()`. Debounce code has nothing to do with `attachInterrupt()`.

I suppose an alternative that doesn't add any significant irrelevant code to the example would be to set the LED pin to mirror the state of the interrupt pin:
```c++
const byte ledPin = 13;
const byte interruptPin = 2;
volatile byte state = HIGH;

void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
}

void loop() {
  digitalWrite(ledPin, state);
}

void blink() {
  state = digitalRead(interruptPin);
}
```
That will not be visibly affected by contact bounce, but it's not a very interesting demonstration of the use of interrupts.

PaulStoffregen, 2019-03-27:
Perhaps the attachInterrupt page could mention the Bounce library and specifically recommend *not* using interrupts for pushbuttons, switches and other mechanisms that suffer from mechanical chatter?  This seems to be a very common misunderstanding among novices.  The documentation could really do much better to guide users towards the proper approach to meet their needs.

PaulStoffregen, 2019-03-28:
I submitted a pull request.  Hopefully it will help others avoid this common misunderstanding.

https://github.com/arduino/reference-en/pull/576

animeshsrivastava24, 2019-06-05:
@per1234 Along with this Issue, I am trying out all the examples mentioned on the Reference Pages with help of my Arduino Uno Board and other specifications necessary and will be happy to report any issue with any present example that will need an update or correction.