arduino/reference-en #588: Is the syntax for `Stream.parseInt()` correct?

SeppPenner, 2019-04-28:
[image]

https://www.arduino.cc/reference/en/language/functions/communication/stream/streamparseint/

I do not get the syntax there? What do the `'` do there??

closed 2019-05-01 by robsoncouto

SeppPenner, 2019-05-03:
@robsoncouto Thank you for the merge, I was not completely sure if this was a syntax error or not.

per1234, 2019-05-03:
Not only was the markup and formatting of the syntax section wrong, the information in the syntax section was also completely wrong. Investigating and correcting that issue lead me to find the same false information on three other pages. Thanks for bringing this to my attention @SeppPenner!

SeppPenner, 2019-05-03:
@per1234 Oh, I didn't see that. But glad to know 👍