arduino/reference-en #591: Add reference links from the language website (e.g. German) to the orginal website?

SeppPenner, 2019-05-01:
After some suggestions in the forum (http://forum.arduino.cc/index.php?topic=612980, in German), I wanted to ask if it is possible to add links on the language website (e.g. https://www.arduino.cc/reference/de/) to the original english website (e.g. on bottom of the page) to have a direct reference to the original. I know that I can just add the links on each page but as this is a bigger change, I wanted to ask the community and the arduino members first.

This was a suggestion from the forum, not mine.

SeppPenner, 2019-05-03:
Another suggestion was to maybe add a language switch to easily switch from e.g. English to German like Microsoft has on their pages (e.g. https://docs.microsoft.com/de-de/ef/core/managing-schemas/migrations/). I do not know how much effort this would be.

per1234, 2019-05-03:
I'd like to see some way of switching to any of the available languages on the reference pages. I think this is commonly done on other major websites with little flag icons. [Arduino's Software page](https://www.arduino.cc/en/Main/Software) uses a dropdown menu to select language:

[image]

So I suppose that would be the way to go to follow an existing standard.

As far as I know, the availability of translations of the Arduino Language Reference are not currently advertised in any way (or does arduino.cc automatically selects the correct language?). Having a language menu would make it more obvious that this resource is available to the users.

smellai, 2019-05-07:
https://github.com/bcmi-labs/reference/issues/63

SeppPenner, 2020-04-28:
This is done, isn't it?

[image]

closed 2020-04-28 by SeppPenner

per1234, 2020-04-28:
I think we can consider it done. Not all the languages have been added, but this is because their translations are not completed yet. So the infrastructure is all in place, it's only a matter of volunteers stepping in to help get us to full translation of the reference.