arduino/reference-en #593: [Question to the community]: Add the possibility to download the reference as PDF (or EPUB)?

SeppPenner, 2019-05-07:
From the German forum (See https://forum.arduino.cc/index.php?topic=612980.20), I got the question if the reference is available as PDF (or EPUB) download for offline availability.

As far as I know, this is not yet possible. Does it make sense to add this as a feature?

SeppPenner, 2019-05-07:
Maybe something like this https://github.com/asciidoctor/asciidoctor-pdf could help here...

per1234, 2019-05-07:
A forum member made a PDF version of the Arduino Language Reference some years ago and there was a good deal of interest in it:
https://forum.arduino.cc/index.php?topic=392195.0
The problem was that the Language Reference content is regularly being added to and improved, and that person only made the PDF once, so their PDF quickly became outdated. It was frustrating to see people complain about documentation problems that had been fixed long ago in the online Language Reference.

If this was to be done by Arduino, I think it's very important for it to be set up to update automatically so that the PDF version is always in sync with the latest Language Reference content.

>Maybe something like this https://github.com/asciidoctor/asciidoctor-pdf could help here...

I've been working with asciidoctor-pdf a lot lately and it's really great.

SeppPenner, 2019-05-07:
@per1234 I totally agree with you. Automatic updates need to be there. I have not yet worked with asciidoctor-pdf yet, but good to know that it's great 👍

SeppPenner, 2019-05-08:
I doesn't look that bad... The headings are not formatted properly though.

[noTone.pdf](https://github.com/arduino/reference-en/files/3158595/noTone.pdf)

SeppPenner, 2019-05-08:
It doesn't look that bad. However, the headings look strange...

[noTone.pdf](https://github.com/arduino/reference-en/files/3158606/noTone.pdf)

SeppPenner, 2019-05-08:
It doesn't look that bad. However, the headings look strange...

https://filehorst.de/d/cwbjkEaA

SeppPenner, 2019-05-08:
Test

SeppPenner, 2019-05-08:
Something's wrong here...

[image]

SeppPenner, 2019-06-19:
Maybe this is also worth a look: https://github.com/russcam/asciidocnet (Depending on which languages are supported with the arduino build scripts).