arduino/reference-en #600: I want translate https://www.arduino.cc/reference/en/ into Polish.

tomb0x, 2019-05-24:
I want translate https://www.arduino.cc/reference/en/ into Polish language. I am one of the translators in the Arduino 1.5 IDE transifex team.

per1234, 2019-05-24:
@gvarisco please create https://github.com/arduino/reference-pl and give me write access to it. Thanks!

tomb0x, 2019-05-26:
>          github.com @gvarisco  please create  github.com github.com  and give me write access to it. Thanks!

How do I create a reference-en in arduino but not in tomb0x?   I created  github.com github.com  and what next?   I&#39;m new to github ... please help. Greetings.

per1234, 2019-05-26:
Hi @tomb0x thanks so much for your interest in helping with the Arduino Language Reference translation project!

We create a dedicated repository for each translation language as volunteers step forward and offer their assistance in translating. We didn't already have a repository for the Polish language translation so we need to create one. It will be at https://github.com/arduino/reference-pl. After that, I'll copy over the English language content files from this repository. Each of those files will need to be translated to Polish. As soon as we have the repository set up, I'll let you know and provide any assistance you might need.

If you want to get started translating right away, you can go ahead and do so on your own repository or computer and then just copy those files over to the reference-pl repository once it's created.

gvarisco, 2019-05-27:
> @gvarisco please create https://github.com/arduino/reference-pl and give me write access to it. Thanks!

done!

SeppPenner, 2019-06-08:
Hi @per1234. What's the status? Any issues with https://github.com/arduino/reference-pl/pull/1?

per1234, 2019-06-08:
@SeppPenner there are no issues with https://github.com/arduino/reference-pl/pull/1. It's only waiting for one of my colleagues from Arduino to merge it. Once that's done, the translation work can proceed.

gvarisco, 2019-06-11:
Done, sorry for the delay!

closed 2019-06-11 by gvarisco