arduino/reference-en #601: Difference between `String` and `string` in the documentation?

SeppPenner, 2019-05-24:
Is there a difference between `String` and `string` in the Arduino language or does `string` even exist? I'm asking because I fell over this in the file https://github.com/arduino/reference-en/blob/master/Language/Variables/Data%20Types/String/Functions/concat.adoc. There are both versions in the documentation:

[image]

per1234, 2019-05-24:
Yes, there is a difference. String is a class intended to make it easier to work with text. It does dynamic memory allocation and has a lot of overhead so you have to be very careful when using it:
https://www.arduino.cc/reference/en/language/variables/data-types/stringobject/

string is a null terminated char array:
https://www.arduino.cc/reference/en/language/variables/data-types/string/

The two names being so similar causes a lot of confusion. It's very important that we make sure to always use `String` when talking about the former and `string` when talking about the latter.

closed 2019-05-24 by per1234

SeppPenner, 2019-05-24:
Ok, I didn't know how this is treated in the Arduino language. I just know the same thing from C#. There however, the difference is not really big. Thanks for the explanation, I thought it was a bug :D