arduino/reference-en #602: How do we handle constants formatting in the reference?

SeppPenner, 2019-05-26:
Check e.g. https://www.arduino.cc/reference/de/language/functions/advanced-io/pulsein/. For example HIGH and LOW are written in different formattings: As code, as link or big. How do we want to format this as a standard?

per1234, 2019-05-27:
This is similar to https://github.com/arduino/reference-en/issues/462.

My opinion is that whenever code words are used in text, they should be wrapped in backticks. I'm fine with also linkifying those code words.

In cases where the link is important (rather than a "convenience", "linkify everything" link), for example:
>Please see the digitalWrite() reference page

Things are more complicated. The problem is that link text in backticks is not colored as a link on arduino.cc. The only way for someone to know that text is a link is if they happen to hover their mouse over the text and notice the pointer changes. There is a screenshot that shows this here: https://github.com/arduino/reference-en/issues/462#issuecomment-427698107.

So in the case of an important link with link text that is exclusively a code word, I think it would be best to not use backticks.

SeppPenner, 2019-05-27:
> My opinion is that whenever code words are used in text, they should be wrapped in backticks. I'm fine with also linkifying those code words.

I would linkify all of them. I know, it's huge effort though...

> Things are more complicated. The problem is that link text in backticks is not colored as a link on arduino.cc.

I have seen that as a problem somewhere, too...

> So in the case of an important link with link text that is exclusively a code word, I think it would be best to not use backticks.

I totally agree with that.

SeppPenner, 2019-06-08:
So: If the code word is used, put it in backticks. If it's important, link it and do not use backticks. I will just handle things like this in the future.

closed 2019-06-08 by SeppPenner