arduino/reference-en #610: Documenting the Keyboard Modifiers

animeshsrivastava24, 2019-06-02:
The [Keyboard Modifier](https://www.arduino.cc/reference/en/language/functions/usb/keyboard/keyboardmodifiers/) only shows the Leonardo’s definitions for modifier keys, variations for other boards can also be added.

per1234, 2019-06-02:
Please provide an example of a keyboard modifier for another board which is different from those listed on that page.

animeshsrivastava24, 2019-06-02:
While viewing the Keyboard Modifiers, I landed on https://forum.arduino.cc/index.php?topic=297878.0 and https://github.com/SFE-Chris/UNO-HIDKeyboard-Library , there were discussions on the Keyboard Key Codes. Please guide, are the discussions on these forum, relevant to this issue.

per1234, 2019-06-02:
>I landed on https://forum.arduino.cc/index.php?topic=297878.0

That is outdated. Years ago, the Keyboard and Mouse functionality was in the core library but luckily that code was split out into libraries which are bundled with the Arduino IDE and shared by all the boards with these capabilities. So, as far as I know, the same keyboard modifiers apply to all boards:
https://github.com/arduino-libraries/Keyboard/blob/master/src/Keyboard.h#L37-L84

>and https://github.com/SFE-Chris/UNO-HIDKeyboard-Library

That's a 3rd party library, so not relevant.

---
I do notice a couple of serious problems after a quick glance at the Keyboard Modifiers page:
- "The Leonardo’s definitions for modifier keys are listed below"
  - The "Leonardo" part of that sentence is misleading. That sentence should be changed to "The definitions for modifier keys are listed below:"
- The table is missing the definitions for F13-F24

@animeshsrivastava24 if you would like to submit pull requests to fix the issues listed above, I would be happy to review and merge them.

animeshsrivastava24, 2019-06-02:
@per1234  Yes, definitely I will work on it and fix the issue.
Also, I request you to please help me out with the review of the mail. I have been waiting eagerly for your reply, it will be a great help and help me to summarize more topics for GSoD'19.

per1234, 2019-06-02:
>I request you to please help me out with the review of the mail. I have been waiting eagerly for your reply, it will be a great help and help me to summarize more topics for GSoD'19.

I'm sorry for the delay responding to your emails. There is some internal discussion regarding whether I am permitted to respond to these emails now that GSoD's "Technical writer exploration" phase has ended. I am waiting to hear a definitive decision on this (which was delayed due to the weekend).

animeshsrivastava24, 2019-06-02:
@per1234  Thank you so much for the response. So, will it be possible to take guidance from the mentors regarding the content proposal or not. Sure, I'll wait further and in meanwhile, keep contributing and improving the Documentation.

per1234, 2019-06-03:
Fixed by https://github.com/arduino/reference-en/pull/611

closed 2019-06-03 by per1234