arduino/reference-en #617: Translation of Arduino Reference to the Hindi Language

animeshsrivastava24, 2019-06-08:
Please create a repository for translation to the Hindi Language.

per1234, 2019-06-08:
@animeshsrivastava24 do you want this repository created for a Season of Docs project, or is it for volunteer work uninvolved with the Season of Docs program?

animeshsrivastava24, 2019-06-08:
@per1234  Thank you so much for the response. I'll love to contribute as a volunteer for the work.

per1234, 2019-06-08:
@gvarisco please create the https://github.com/arduino/reference-hi repository and give me write permissions in it. Thanks!

gvarisco, 2019-06-11:
Done, https://github.com/arduino/reference-hi has been created.

per1234, 2019-06-12:
Thanks @gvarisco!

@animeshsrivastava24 the reference-hi repository is now populated with the English Language Reference content awaiting translation to Hindi.

closed 2019-06-12 by per1234