arduino/reference-en #620: missing documentation about Serial.write() / Serial.print();

dsyleixa, 2019-06-09:
as to the  documentation of Serial.write() / Serial.print(); 
it is not made clear what happens if one has to send a msg cstring which is very large (e.g. 1024 bytes) and thus always (probably) exceeds the Serial buffer size. Hints about length and buffer sizes only refer to IDE 1.0, nothing about 1.8.8 or 1.8.9.:
in https://www.arduino.cc/reference/en/language/functions/communication/serial/write/#howtouse
> As of Arduino IDE 1.0, serial transmission is asynchronous. If there is enough empty space in the transmit buffer, Serial.write() will return before any characters are transmitted over serial. If the transmit buffer is full then Serial.write() will block until there is enough space in the buffer. To avoid blocking calls to Serial.write(), you can first check the amount of free space in the transmit buffer using availableForWrite().

Here is also no advice available how to proceed, if the msg is ALWAYS far bigger than e.g. 64 bytes and has to be transmitted in a whole.

matthijskooijman, 2019-06-09:
> As of Arduino IDE 1.0,

I believe this is intended to be read as "In 1.0 and all newer versions", so the paragraph you quoted applies to 1.8.x as well. Does that provide sufficient information, or are you still missing something?

dsyleixa, 2019-06-09:
thanks for your reply,
yes, indeed I am missing information about:
 how to proceed exactly, if the msg is ALWAYS far bigger than e.g. 64 bytes and has to be transmitted in a whole (compileable example code)

matthijskooijman, 2019-06-09:
If you just pass the entire 1024-byte buffer to `Serial.write()`, it should be sent as single block with all bytes back-to-back (provided that interrupts are never disabled for > 1 byte time). However, the call to `write()` will block until 1024-64 bytes are sent, the remaining 64 bytes will be sent in the background. Note that if you split up your data over multipler `write()` calls, the bytes will still be sent back-to-back, since you can prepare the next part of the data in the time the 64-byte buffer is being emptied. As long as you keep the buffer non-empty, bytes will be sent back-to-back.

I suspect that the problem you're seeing is not related to Serial buffer sizes or blocking, but something else in your sketch. Are you sure you have enough memory available on your device? You're allocating 2k of memory in your arrays, but e.g. the Uno only has 2k of memory. In any case, this is not the right place for getting support with your code, so I suggest you ask elsewhere (e.g. the forum or IRC).

Getting to back to the main topic of this issue: Is there anything you think should be improved on the documentation? If so, do you have specific suggestions? If not, I think this issue should be closed.

dsyleixa, 2019-06-09:
I am using the Mega2560 and optionally the Due.
I cannot send in several blocks because of handshake send/receive with the partner MCU.

ut tbh, I still don't the information about 

> However, the call to write() will block until 1024-64 bytes are sent, the remaining 64 bytes will be sent in the background. 

Does that mean, if I was sending not 1024 but just 64 bytes less, i.e. in a  960 bytes array, taht then all the 960 bytes will be sent all at once in a whole, no obscured or remaining bytes vanishing in the background?
The partner MCU reads until '\n' which can be located at either , e.g. at position 10, pos 200, or pos 900; but if Serial.write() is blocking before that position, then the entire message is vanished, either if in the forgroung or in the background

This behaviour about big arrays sent by  Serial.write (actually I am using Serial.print() ) should be better explained in the documentation, too.

matthijskooijman, 2019-06-09:
I'm not sure what you're asking exactly. Perhaps you're misunderstanding what "blocking" means. In this case, "blocking" means that the `write()` function does not return immediately, but has to wait until some data is sent over the serial line. This does not mean that the serial output is blocked or stopped (on the contrary - the `write()` function blocks (waits) while the serial port is sending out data to make room in the buffer.

With `back-to-back` I mean that serial bytes are sent (out of the serial port) directly after each other, without significant gaps (which I gather is what you need).

> The partner MCU reads until '\n' which can be located at either position , e.g. at position 10, pos 200, or pos 900; but if Serial.write() is blocking before that '\n' position, then the entire message is dropped and vanishes, either if in the forground or in the background or if some more are following somehow.

This suggests that you might not be doing proper framing on the receiving side. Typically, a serial stream should be seen as exactly that: A stream of bytes. Calling `write()` adds bytes to the stream, and `read()` on the other end removes bytes from that stream. Serial lines are not, by themselves, packet-based in the sense that you cannot see on the receiving end where one `write()` call ends and the next one starts. The only thing you can see, if you write your code for that, is how much delay there is between characters on the serial line. This means that if you want to delineate your serial data stream into packets, you have to use some framing mechanism (e.g. by terminating each packet using a `\n` *and* ensure that `\n` does not occur anywhere inside the packet, or by detecting a delay between characters, though that is harder to make reliable).

In any case, here's another attempt to explain how serial writing works. Perhaps this helps you to understand what you need and/or what is happening exactly.
 - The `Serial` object contains an internal buffer of 64 bytes.
 - Whenever there is data in the buffer, an interrupt handler make sure that the data in the buffer is being sent through the TX pin. Whenever one byte is completed, the next byte starts immediately.
 - The `write()` function adds data into the buffer. When the buffer has insufficient free space, `write()` fills up the buffer entirely and then waits (blocks) for the background interrupt handler to send bytes and free up space in the buffer again. Whenever the interrupt handler sends and frees up one byte, `write()` adds a new byte to the buffer. This continues until all data passed to `write()` has been put into the buffer, after which `write()` returns (and the 64 bytes still in the buffer will be sent in the background after that).

This means that if you pass data to `write()` and it returns, up to 64 bytes will be transmitted in the background. If you call `write()` again before the buffer is empty, both chunks of data will be sent back-to-back and on the outside there is no way to tell the different between both writes (iow, calling `write()` two times with e.g. 64 bytes close together looks the same on the outside as calling it once with 128 bytes).

If you need to know when data is actually sent out on the serial port (not just when it is put in the buffer for later transmission), you can use `Serial.flush()` which waits (blocks) until the buffer is empty and all data was sent.

dsyleixa, 2019-06-09:
thank you, that kind of information should be added to the documentation.
But additionally:
Actually all that buffer thing should be always handled correctly by Serial, no matter if I pass 40, 64, 200, 900, or 1024 bytes, and tbh, I do still not understand what frees when and by which condition or not, the information is still sort of confusing and ambiguous, especially about this background thing.
In easy words:
If I send 1024 0r 960 or 900 bytes of an array, I want to be sure that all of that always will be sent completely.
What does one have to do or to cede to ensure that? (see my code snippet above)

Nonetheless, the Serial.flush() again makes that additionally more  ambiguous:
at which place should that be added in the code above?
Also a reasonable documentation about that flush thing is missing.

So after all, I cannot become clear about all that reading the current Serial documentation.

If you wish, I can also provide the entire compileable code for both UART partners (if that might help)

matthijskooijman, 2019-06-09:
> If I send 1024 0r 960 or 900 bytes of an array, I want to be sure that all of that always will be sent completely.

This is what `Serial.write()` should do. Any data you pass to it, will be sent. All the background and buffer stuff is essentially just implementation details that should not even be entirely relevant (they only affect timing, but should never affect what is being sent). In this case, I suspect there is some problem in your sketch (maybe your buffers are too small?) instead.

> Also a reasonable documentation about that flush thing is missing.
It should be documented at https://www.arduino.cc/en/Serial/Flush, though the docs are indeed short.

> If you wish, I can also provide the entire compileable code for both UART partners (if that might help)

I think that would help, but as I said, this is not the right place (nor do I have time to dive into this further). I suggest you post your code to the forums and ask for help there. In the meanwhile, I think this issue can be closed. If you later have specific suggestions on how the documentation can be improved, feel free to comment here again.

closed 2019-06-09 by per1234

dsyleixa, 2019-06-09:
if the documentation was complete, clear, widly explained and not confusing as it is right now, that would be sufficient, but meanwhile it's not, neither for read, write, available or flush.
So this issue is not resolved at all, pleas reopen it until a reasonable and complete and non-ambiguous documentaion is available.

The Arduinocc forum is no place for me any more to post anything at all because of accusing, offending, and insulting trolls - I'm fed up with that.

per1234, 2019-06-09:
>please reopen it until a reasonable and complete and non-ambiguous documentaion is available.

The documentation seems fine as it is to me. If you want to provide specific suggestions for how it can be improved then we can act on that. Otherwise, there is nothing we can do and so the issue is not helpful.

>Not even a documentation about flush is provieded

Serial.flush() is indeed documented:
https://www.arduino.cc/en/Serial/Flush

>The Arduinocc forum is no place for me any more to post anything at all because of accusing, offending, and insulting trolls - I'm fed up with that.

I'm sorry you feel that way, but this is not an appropriate place to ask for help with your project. This is a place to report bugs or make specific suggestions for improvements. You'll need to find somewhere else to ask for help with your project.

dsyleixa, 2019-06-09:
It's about a missing, reasonable documentation for common users, of course advanced people like you understand already everything so it mistakenly seems to be sufficient (for YOUR skills).
For common users it's definitely not, 
Also ss to flush, as stated, it's rediculous, not even provideing an example code where and when to use it.

dsyleixa, 2019-06-09:
Anyway, as long as the documentary remains so poorly, at least it does not help to write reasonable or even optimal program code...!
Please reopen until it's fixed!