arduino/reference-en #632: "See Also" links for Variables

animeshsrivastava24, 2019-06-23:
The following files miss the content in "See Also" Section under Variables:
1.`reference-en/Language/Variables/Variable Scope & Qualifiers/` 
* scope.adoc 
* static.adoc

2.`reference-en/Language/Variables/Utilities/` 
* sizeof.adoc

3.`reference-en/Language/Variables/Data Types/` 
* word.adoc 
* string.adoc 
* size_t.adoc 
* float.adoc 
* double.adoc 
* boolean.adoc

4.`reference-en/Language/Variables/Constants/` 
* constants.adoc 
* floatingPointConstants.adoc 
* integerConstants.adoc

It is a reference to issue #487

per1234, 2019-06-24:
I just checked all these and the automated generation of "See Also" links are working fine for me. @animeshsrivastava24 would you mind posting a screenshot of the problem as it appears for you on one of these pages?

animeshsrivastava24, 2019-06-24:
@per1234  I went through these pages and found that below "See Also" the contents are missing. 
For example:
https://github.com/arduino/reference-en/blob/master/Language/Variables/Utilities/PROGMEM.adoc
In this, the **See Also** Section has
[image]

but in https://github.com/arduino/reference-en/blob/master/Language/Variables/Utilities/sizeof.adoc
The **See Also** section has nothing below it.
[image]

I went through all these pages and found these missing contents.
So, please guide me, is this an issue or not?
We never had a discussion over this topic so I might be having little to no information about it. It will be valuable for me.

per1234, 2019-06-24:
During the website build process, links are automatically added to the See Also section for all pages in the same folder. You can't see these on GitHub, but you will see them on arduino.cc So if you look at the actual sizeof reference page:
https://www.arduino.cc/reference/en/language/variables/utilities/sizeof/
you'll see this:

[image]

---
You can also manually add additional See also links to relevant content outside the folder of the reference page, as was done in the PROGMEM page:
https://github.com/arduino/reference-en/blob/c07a147f049a220853b612763cee476a3194063d/Language/Variables/Utilities/PROGMEM.adoc#L205-L215
the automatically generated links will be added in addition to these during the site build process:
https://www.arduino.cc/reference/en/language/variables/utilities/progmem/

[image]

---
So everything is fine with the See also sections of these pages already.

We should not add See also links for the pages that will already have links added automatically because the generated links adapt to any changes to the content and thus don't impose a maintenance burden.

closed 2019-06-24 by per1234

animeshsrivastava24, 2019-06-24:
Thank you so much @per1234  for helping me understand this.