arduino/reference-en #654: Link Redirect Problem in the Stream Related Files, occurring in the Documentation

animeshsrivastava24, 2019-07-03:
The files present in the Folder: `reference-en/Language/Functions/Communication/Stream/`
Contain Links in each documentation which doesn’t lead us to correct page.
For example:
1. Consider the file [streamAvailable]( https://github.com/arduino/reference-en/blob/master/Language/Functions/Communication/Stream/streamAvailable.adoc)
On line 18: It has a link Stream class which redirects a user to https://github.com/arduino/reference-en/blob/master/Language/Functions/stream, but the correct link should be https://github.com/arduino/reference-en/tree/master/Language/Functions/Communication/Stream 
2. Consider the file [streamFind](https://github.com/arduino/reference-en/blob/master/Language/Functions/Communication/Stream/streamFind.adoc)
On line 17: It has a link Stream.setTimeout() which redirects a user to https://github.com/arduino/reference-en/blob/master/Language/Functions/Communication/streamsettimeout, but it should be https://github.com/arduino/reference-en/blob/master/Language/Functions/Communication/Stream/streamSetTimeout.adoc 

I found these instances in various files of Stream. I checked this on the website and it is working fine there but on the documentation present in the repo. I am a little confused due to this.
@per1234 Please guide me whether this is a problem or not.

per1234, 2019-07-04:
This is because on arduino.cc each page is a folder, while on GitHub each page is only a file. You will notice this in the relative links where there is one more layer of .. than you would expect. So there is no problem. The files in the reference repositories are intended to be used to generate the Language Reference on arduino.cc. The fact that GitHub renders asciidoc is convenient, but it should not be considered an indication that the language reference content is intended to ever be read by users here on GitHub.

closed 2019-07-04 by per1234

animeshsrivastava24, 2019-07-04:
Thank you so much for explaining me about it.