arduino/reference-en #704: Behavior not documented: Stream.find() consumes the bytes and also the matched byte.

yck011522, 2019-09-13:
I'm refering to this function: https://www.arduino.cc/reference/en/language/functions/communication/stream/streamfind/

The behavior of the function `Stream.find()` is that it will consume every single byte in the inBuffer  until it fins the matching target.

This is not documented nor expected from the name find(). One can expect a `find()` function to behave similar to `peek()` but it is not.

per1234, 2019-09-13:
Hi @yck011522 I think that would be a good thing to document. Would you like to submit a pull request to make this improvement?

yck011522, 2019-09-14:
Hi @per1234. Glad you agree with me. Sorry I do not have the time now. Perhaps some other people is able give it a try.

per1234, 2019-09-14:
I understand, no problem. Your suggestion is valuable in itself. I'm sure someone will act on this in the near future.

Thanks! Per