arduino/reference-en #713: Internal Analog reference giving wrong analogRead results when external voltage is connected at Aref pin through a 5k resistor

vineethnair1991, 2019-12-02:
Observed a 0.4v difference between the actual analog voltage and the voltage read with Internal Analog Reference 1.1v, when an external analog reference voltage is connected to the AREF pin of Arduino Uno through a 5k resistor. This difference is not seen, when the external voltage to the Aref pin is disconnected.

For MKR1010 its working properly with Internal reference voltages: 1,1.65,2.23,3.3v.

The issue is seen with Mega and Uno boards when an external analog reference voltage was connected to the AREF pin through a 5k resistor for protection, and the internal 1.1volt was used as the analog reference. The circuit was connected as per the note which was given under Arduino's documentation page for analog reference https://www.arduino.cc/reference/en/language/functions/analog-io/analogreference/

per1234, 2019-12-07:
I can reproduce the issue, but I don't know what the best resolution would be. 

Should we just remove that section from the documentation?:
>Alternatively, you can connect the external reference voltage to the AREF pin through a 5K resistor, allowing you to switch between external and internal reference voltages. Note that the resistor will alter the voltage that gets used as the reference because there is an internal 32K resistor on the AREF pin. The two act as a voltage divider, so, for example, 2.5V applied through the resistor will yield 2.5 * 32 / (32 + 5) = ~2.2V at the AREF pin.

Anyone have any input on this?

vineethnair1991, 2019-12-12:
The above section in documentation is still relevant for MKR boards and maybe other boards as well. The issue is only seen for Uno and Mega boards. So we can call out regarding this issue in the documentation for these two Arduino boards and other boards as well after testing.

Removing the section won't be a good idea, specifically for users who want to switch between external and internal reference voltage sources. Also, it would be a good idea to have a protection resistor connected as mentioned in the documentation section because, by default, all Arduino boards would be using default analog reference voltage, and if user makes a connection for external reference voltage before burning the corresponding code to set analog reference as EXTERNAL, there could be potential damage to the board.

Another issue in the documentation is that it mentions there is an internal 32K resistor on the AREF pin of the Arduino board. This is not true since the internal resistance at AREF pin changes for different boards. Probably that's the reason why the same 5k resistor can't be used for all boards.