arduino/reference-en #722: Website not current?

rmlearney-digicatapult, 2020-03-21:
I'm looking for information about attachInterrupt() on the new Nano BLE modules.

I came across the following PR which was merged back in December 2019:

https://github.com/arduino/reference-en/pull/714

But when I go to the actual website as of today:

https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/

There's no mention of the Nano BLE.

Which is the canonical source of information - the Github or the Website? Why aren't the two synchronised?

per1234, 2020-03-22:
Thanks for reporting this @rmlearney-digicatapult.

It looks like the last commit that was published to the website is https://github.com/arduino/reference-en/commit/1e29ea8978eefeca21fa0add7d064e57b862ea9f. The next one, https://github.com/arduino/reference-en/commit/807621719690296e97e45385f68cc63e7cdb5b83 was not published.

rmlearney-digicatapult, 2020-03-23:
@per1234 and @matteosuppo I think it should also say that for the mbedOS boards, only RISING and FALLING options work, not CHANGE or HIGH

per1234, 2020-03-23:
@rmlearney-digicatapult good point! Would you like to submit a pull request for this change?