arduino/reference-en #729: Inconsistent Links in const.adoc

omshinde, 2020-04-26:
The links in [https://www.arduino.cc/reference/en/language/variables/variable-scope-qualifiers/const](https://www.arduino.cc/reference/en/language/variables/variable-scope-qualifiers/const) available at [const.adoc](https://github.com/arduino/reference-en/blob/master/Language/Variables/Variable%20Scope%20%26%20Qualifiers/const.adoc) are not redirecting to correct references.

per1234, 2020-04-27:
Thanks very much for reporting this @omshinde! This was a tricky one because at first the links were working fine for me, then they started working and not working intermittently as I investigated. Finally I realized that the links work fine if there is a trailing slash on the URL of the page the link is on, but they are broken if the trailing slash is missing.

For example, if you open this URL:
https://www.arduino.cc/reference/en/language/variables/variable-scope-qualifiers/const/
and click the "`#define`" link, it works fine.

But if you open this URL:
https://www.arduino.cc/reference/en/language/variables/variable-scope-qualifiers/const
(note the missing trailing slash)
and click the "`#define`" link, it's broken.

So the solution you proposed of changing the links in https://github.com/arduino/reference-en/pull/730 and https://github.com/arduino/reference-en/pull/732, as well as @abbaco in https://github.com/arduino/reference-en/pull/727 is not going to work, because it fixes the links for one URL style, while breaking them for the other.

I have reported this issue to Arduino's developers. We'll see if there is a solution that will work with the existing links.

omshinde, 2020-04-27:
Hi @per1234 : Thank you so much for taking this up. Yes, you are correct, the links works like charm with the trailing slash. 

> So the solution you proposed of changing the links in #730 and #732, as well as @abbaco in #727 is not going to work, because it fixes the links for one URL style, while breaking them for the [...]

Yes, I understand. I misunderstood it as a path-location issue, accepting it might dis-orient the complete doc structure. Thanks for pointing it out. Shall I close the issue and PR?

per1234, 2020-04-27:
>I misunderstood it as a path-location issue

I would have thought the exact same thing if I hadn't chanced to access the page with the trailing slash on the URL!

>Shall I close the issue and PR?

We can wait to see what happens with the developers resolving the issue.

tedz2usa, 2020-05-03:
Same issue on the Serial documentation page.

The function links at the bottom of the page don't work here: https://www.arduino.cc/reference/en/language/functions/communication/serial

But adding the trailing slash,
https://www.arduino.cc/reference/en/language/functions/communication/serial/

made the links work.

It seems like there is an issue with the auto-generated relative links from the AsciiDoc format.  I.e., the relative link definition `link:../serial/available[available()]` is not being generated into a correct link.

per1234, 2020-08-17:
The URLs without trailing slash are now automatically redirected to the URL with trailing slash so these links are now working as is. So I'm going to close this, but I really appreciate you both taking the time to bring this issue to our attention!

If you find that the problem has returned, please comment here and I'll investigate.

Thanks!

closed 2020-08-17 by per1234