arduino/reference-en #766: long - error and poor example

MalcolmBoura, 2020-08-10:
https://www.arduino.cc/reference/en/language/variables/data-types/long/

"If doing math with integers at least one of the numbers must be followed by an L"
should read
"If doing math with integers at least one of the values must be of type long, either an integer constant followed by an L or a variable of type long".

The example value is a poor choice for an international science, technology and education resource. Most of the world does not use miles. I suggest "long speedOfLight = 300000L;" or even better "long speedOfLight_km_s = 300000L;"

"Parameters" is an odd choice of subheading. "Description" would be much better.

kengdahl, 2020-08-18:
I implemented your suggestions

kengdahl, 2020-08-18:
Thanks for the good work

closed 2020-08-18 by kengdahl