arduino/reference-en #779: reference-en/Language/Functions/Zero, Due, MKR Family/analogReadResolution.adoc

TimMathias, 2020-09-17:
In the section "Notes and Warnings" there is ambiguous wording:

> For example: using the Due with `analogReadResolution(16)` will give you an approximated 16-bit number with the first 12 bits containing the real ADC reading and the last 4 bits **padded with zeros**.

Consider revising it to:

> For example: using the Due with `analogReadResolution(16)` will give you an approximated 16-bit number with the most significant 12 bits containing the real ADC reading and the least significant 4 bits **padded with zeros**.

This would make it consistent with the wording of the following paragraph:

> If you set the `analogReadResolution()` value to a value lower than your board’s capabilities, the extra least significant bits read from the ADC will be **discarded**.

kengdahl, 2020-09-23:
@dcuartielles I agree with him. Shall I make the changes?

TimMathias, 2020-10-03:
@kengdahl and @dcuartielles, yes, go on, what are you waiting for?

kengdahl, 2020-10-08:
@TimMathias I was asking @dcuartielles about this. As soon as I get the ok from him I'll gladly make the changes.